roboticslab-uc3m/vision #6: [idea] Small QR detector module.

jgvictores, 2015-11-28:
Similar to cv1 or cvFaces, develop cvQR.

PeterBowman, 2018-10-27:
In asrob-uc3m/robotDevastation, we adopted a well-established abandonware library, ZBar: [ref, look for the `ProcessorImageEventListener` class](https://github.com/asrob-uc3m/robotDevastation/tree/5b4a0e81c17f4ba5fb6cd2557a422c2c44338dbd/src/libraries/ImageLib). We also host the Windows 64-bit binaries: https://github.com/asrob-uc3m/ZBar.

jgvictores, 2019-12-04:
Blocked by https://github.com/roboticslab-uc3m/vision/issues/79

PeterBowman, 2020-07-08:
OpenCV 4.0 introduces QR code handling ([ref](https://github.com/opencv/opencv/wiki/ChangeLog#version400), https://github.com/asrob-uc3m/robotDevastation/issues/78#issuecomment-654759518):

> QR code detector and decoder have been added to opencv/objdetect module together with live sample. Currently, the decoder is built on top of QUirc library (which snapshot we put into opencv). Thanks to our intern Alexander Nesterov for this important contribution!

jgvictores, 2020-11-01:
Unblocked, it should now be rather straightforward to implement this.

PeterBowman, 2020-11-02:
I'm eager to work on this. Some research:

* OpenCV 4.0: first implementation ([docs](https://docs.opencv.org/4.0.0/de/dc3/classcv_1_1QRCodeDetector.html))
* OpenCV 4.3: adds support for multiple QRs on image ([docs](https://docs.opencv.org/4.3.0/de/dc3/classcv_1_1QRCodeDetector.html))
* current master, i.e. OpenCV 4.6: adds support for curved QRs ([docs](https://docs.opencv.org/master/de/dc3/classcv_1_1QRCodeDetector.html), https://github.com/opencv/opencv/pull/18003)
* sample application: [*qrcode.cpp*](https://github.com/opencv/opencv/blob/691c3d1e3c86d7fed233301a06a0095b8612d2d8/samples/cpp/qrcode.cpp)

QR code detection is built-in in OpenCV, whereas QR decoding is handled by the [Quirc library](https://github.com/dlbeer/quirc) embedded and conditionally built along with OpenCV (enabled by default).

PeterBowman, 2020-11-03:
Done at https://github.com/roboticslab-uc3m/vision/commit/e62cf5370dd3cf45073386728297b441cd55aa18: C++ code + Python example + unit test. The new `QRDetector` device supports multiple QR codes in the supplied image if compiled with OpenCV 4.3+.

closed 2020-11-03 by PeterBowman

PeterBowman, 2021-04-22:
OpenCV 4.5.2 (contrib) features the [WeChat QR code detector](https://docs.opencv.org/4.5.2/d5/d04/classcv_1_1wechat__qrcode_1_1WeChatQRCode.html) based on CNN models.