roboticslab-uc3m/vision #10: cv1 and cvFaces documentation

jmgarciah, 2016-02-03:
Documentation currently hosted at:
http://robots.uc3m.es/dox-teo-head/group__cvFaces.html
http://robots.uc3m.es/dox-teo-head/group__cv1.html
Lacks information regarding
* Configuration parameters on init
* Input ports and contents
* Output ports and contents

jmgarciah, 2016-02-15:
PS: --noMirror parameter should be documented at doxygen and code level.

rsantos88, 2016-02-18:
Added tables with parameters of cv1 and cv1Faces documented at doxygen in the next pull request: d23a265abde84f78f2e194b366179f4965759fb6

jgvictores, 2016-03-08:
Here's some pointers on a description to be added to what currently is cv1 (may change name! see #22):
It's mainly composed by: classical morphological operations, some kind of binary thresholding, then opencv blobs/contours...

Kuronogard, 2017-10-17:
The links are down.
It would be really useful to include examples in the documentation.

PeterBowman, 2017-10-17:
@Kuronogard fixed (changed `dox-teo-head` to `dox-vision`).

PeterBowman, 2019-06-13:
> Added tables with parameters of cv1 and cv1Faces documented at doxygen in the next pull request: [d23a265](https://github.com/roboticslab-uc3m/vision/commit/d23a265abde84f78f2e194b366179f4965759fb6)

The original requests have been fulfilled, docs may be always improved thereafter - hence closing.

closed 2019-06-13 by PeterBowman