roboticslab-uc3m/vision #11: Define the scope of the repository

David-Estevez, 2016-02-03:
I have seen some commits lately for the "waiter" application on this repo, and I was wondering if we should keep this repo just for generic functionalities (drivers o modules that can be used in several different applications) and create new repositories for specific applications.

What do you think, guys?

jgvictores, 2016-02-03:
Hmmm that's an interesting question, could you also post it on https://github.com/roboticslab-uc3m/teo-body?
Focusing on this repository, regarding computer vision programs, right now we have:
* cv1: performs generic computer vision segmentation through classic algorithms
* cvFaces: similar output, but uses haar cascades for faces.
Promoting generic functionalities, cvFaces (due to the fact that merging with cv1 is a bit more tedious but not impossible) could (should!) be renamed cvHaar and accept any Haar xml and therefore become pretty generic.
In this line of reasoning, I believe a program that provides generic algorithms should stay. A program that only serves for a specific case and has no intention of being generic, well, maybe it makes sense to move it to a different repository within the organization (such as what we do with https://github.com/roboticslab-uc3m/textiles).
So... I'm pretty inclined towards thinking a new repository would be a good option. However, @jmgarciah, I leave it to your decision if you believe the "waiter-head" branch (essentially, cvBottle) should be moved to a new https://github.com/roboticslab-uc3m/waiter repository or similar.

smcdiaz, 2016-02-04:
I think we should discuss about it before making any modification in the organization of the repo

jgvictores, 2016-05-05:
Waiter branch will not be merged. The focus of the following [programs](http://robots.uc3m.es/dox-teo-head/group__teo__head__programs.html) is clear:
* colorRegionDetection
* haarDetection
* speechRecognition.py

However, most probably cv1ToRoot will be renamed, and multipleDump will be renamed or moved to a different repository.

closed 2017-04-22 by jgvictores

jgvictores, 2018-02-13:
Update: multipleDump deleted, see #80

jgvictores, 2018-02-13:
Note: old issue, when this repo was teo-head and also included speech (now at https://github.com/roboticslab-uc3m/speech).