roboticslab-uc3m/vision #13: Avoid "using namespace" to avoid namespace conflicts

David-Estevez, 2016-02-05:
Use alternatives such as:
  * Not using "using namespace"
  * Using named namespaces (similar to Python's "import x as y")

jgvictores, 2016-02-05:
Totally agree.
+1 to Not using "using namespace".

David-Estevez, 2016-02-10:
@rsantos88 cv::vector does the same as std::vector ?? (Commit 0807c9ebd90d30709b6c519760a1b2d619a800b6 )

rsantos88, 2016-02-10:
@David-Estevez in this moment, there are not “using namespace” in teo-head proyect and the code compiles perfectly but is necessary to create a tests system for checking if the programs work well.

David-Estevez, 2016-02-10:
cv::vector might be due to a "using namespace" inside OpenCV, that has added all std:: members to the cv:: namespace.

@rsantos88 cv::vector should be replaced by std::vector in all instances.

rsantos88, 2016-02-10:
@David-Estevez  Now cv::vector has been replaced by std::vector in all instances of teo-head (Commit c950343ed9c23dbf1a839c81225cb0b5cfbcf4c4)

David-Estevez, 2016-02-10:
@jgvictores could you test this refactoring before merging with the main branch?

jgvictores, 2016-02-19:
Tried and merged at 4a34f5e7826b5dae5383c2bbfebd7143447524d2.

closed 2016-02-19 by jgvictores