roboticslab-uc3m/vision #22: New name for cv1

jgvictores, 2016-03-08:
For starters, "1" should be dropped.
It's mainly composed by: classical morphological operations, some kind of binary thresholding, then opencv blobs/contours...
@David-Estevez Any ideas?

David-Estevez, 2016-03-08:
What about "cvUtils" or "cvTools"?

Although the prefix "cv" gives me the impression that the code belongs to the OpenCV main code... "VisionUtils" or "VisionTools" also sound good to me.

jgvictores, 2016-03-08:
+1 to cv -> vision
Utils or Tools sounds more like collections of scripts, how about some of the following or variants of them?
"visionColorBlobs"
"visionColorSegmentation"

David-Estevez, 2016-03-08:
how about "BasicVision" or "Vision"?

jgvictores, 2016-03-08:
I've always liked lowerCamelCase for programs, and category as prefix (even in contrast with standard english :smile:). Ok with "visionBasic"?

This change will not be immediate, but it's good as we converge into a naming standard.

David-Estevez, 2016-03-08:
I thought it was related to some sort of library, so I used CamelCase for it.

Just as a side-note, computer vision literature or at least the book I have at hand) names the different stages in vision using the follow naming:

* Early vision: linear filters, textures, edge detection, stereo matching, etc
* Mid-level vision: segmentation, tracking, etc
* High-level vision: classification and semantics

jgvictores, 2016-03-14:
Final decision:
* cv1 -> colorRegionDetection

jgvictores, 2016-04-04:
Done at https://github.com/roboticslab-uc3m/teo-head/commit/dbf20fec53ba9699558c06adf9c4ff25472f7c0f.

commit dbf20fec (2016-04-04): fix #22
closed 2016-04-04 by jgvictores