roboticslab-uc3m/vision #31: Pointgrey camera too hot

David-Estevez, 2016-06-10:
Pointgrey camera heats a lot even when not in use, which leads to hardware livespan reducción (and other potential problems)

David-Estevez, 2016-06-14:
I am currently making some tests *ex vivo* to determine if is a problem of the camera, its a normal condition of if it is related to the camera environment inside TEO head.

David-Estevez, 2016-06-14:
*In vivo* tests:

*Ex vivo* tests:
 * Camera connected to another computer, outside TEO head: camera gets warm, but not too hot.

smcdiaz, 2017-02-14:
Antes de haber hecho nada lo normal hubiera sido medir la temperatura y consultar las características de la cámara. From the manual:

The Flea2 is an ultra-compact camera. As a result of packing the Flea2 electronics into a small
space, the outer case of the camera can become very warm to the touch when running in some
high data rate video modes. The case can reach temperatures up to 45° Celsius under normal
operating conditions. This is expected behaviour and will not damage the camera electronics.

Temperatura >>> 45º ->instalamos ventilador
Temperatura <= 45º OK!! no es necesaria la ventilación forzada

jgvictores, 2018-01-26:
This issue was moved to roboticslab-uc3m/teo-hardware-issues#8

closed 2018-01-26 by jgvictores