roboticslab-uc3m/vision #32: add new words for speechrecognition

jmgarciah, 2016-06-17:
what should we do for adding new words in the "speechrecognition" fuction ?? where can we find the documentation ???

I am developing the waiter demo, and for me, it is necessary to include the next commands:

- VOCAB_HELLO_TEO
- VOCAB_GO_TEO
- VOCAB_WATER_PLEASE
- VOCAB_STOP_TEO

jgvictores, 2016-06-17:
Collections of words are currently stored in [share/speechRecognition/conf](https://github.com/roboticslab-uc3m/teo-head/tree/a560e0865678d77a0be712f74ad6f211248e39ee/share/speechRecognition/conf).
The .dic and .lm have been generated from a .txt using a tool like [this one](http://www.speech.cs.cmu.edu/tools/lmtool-new.html).

jgvictores, 2016-06-17:
Just to not make many changes to speechRecognition.py, for now I've just updated the dictionary at d3e94034a09aa23991f1d7f94548ead7fd3fbce8.

jgvictores, 2016-06-28:
Just an update on this: the bigger dictionary updated at d3e94034a09aa23991f1d7f94548ead7fd3fbce8 is causing ASR to fail quite a bit. This issue should continue open until we implement a better solution.

rsantos88, 2016-11-04:
Now, we have a [new version of speechRecognition](https://github.com/roboticslab-uc3m/teo-head/blob/develop/programs/speechRecognition/speechRecognition.py), able to change deferents dictionaries by yarp. I have just updated it with separated dictionaries (#47). For now, the dictionaries will be only in english for this reason (see [FAQ page](http://www.speech.cs.cmu.edu/tools/lmtool.html) of lmtool):

`Currently lmtool is configured for the English language (and its American dialect in particular). If you upload a corpus in a different language, your output is likely unpredictable. We are working on this. The current version does not deal gracefully with Unicode; this is also being worked on. `

jgvictores, 2016-11-05:
The [newest version of lmtool](http://www.speech.cs.cmu.edu/tools/lmtool-new.html) does not say this any more. Its corresponding [FAQ now mentions a quicklm solution](http://www.speech.cs.cmu.edu/tools/FAQ.html#_Toc252644378).

rsantos88, 2016-12-21:
Estoy intentando añadir en el directorio share/speechRecognition/conf los diferentes modelos acústicos (hmm), tanto para el inglés como para el español, además de incluir los respectivos diccionarios (.dict y .lm) que se usarán en las demos (ver [aquí](https://github.com/rsantos88/teo-head/tree/spanishLanguageModel/share/speechRecognition/conf)). Quisiera saber cuales son las modificaciones que habría que realizar en el [CMakeList.txt](https://github.com/rsantos88/teo-head/blob/spanishLanguageModel/share/speechRecognition/CMakeLists.txt) para que se copien correctamente estos directorios en /usr/local/share/... 

Gracias!!

jgvictores, 2016-12-21:
Estoy bastante seguro que no hacen falta modificaciones. El CMakeLists al que señalas toma todo .ini, .lm, y .dic que esté dentro de la carpeta conf y los trata por igual (los pasa a contexts con cmake, y a /usr/local/..... con el sudo make install).

jgvictores, 2016-12-21:
Ah, vale, lo has separado por subcarpetas...

rsantos88, 2016-12-21:
Eso es! Más ordenado para separar cada tipo de diccionario... aunque lo importante es separar los modelos acústicos de los diccionarios. Los modelos contienen diferentes tipos de ficheros, por eso imagino que hay que modificar el CMakeLists correctamente.

jgvictores, 2016-12-21:
```cmake
file(GLOB conf ${CMAKE_CURRENT_SOURCE_DIR}/conf/*.ini)
file(GLOB lm_ruta1 ${CMAKE_CURRENT_SOURCE_DIR}/conf/ru/ta1/*.lm)
file(GLOB dic ${CMAKE_CURRENT_SOURCE_DIR}/conf/*.dic)

yarp_install(FILES ${conf} DESTINATION ${TEO-HEAD_CONTEXTS_INSTALL_DIR}/${appname})
yarp_install(FILES ${lm_ruta1} DESTINATION ${TEO-HEAD_CONTEXTS_INSTALL_DIR}/${appname}/ru/ta1)
yarp_install(FILES ${dic} DESTINATION ${TEO-HEAD_CONTEXTS_INSTALL_DIR}/${appname})
```
O algo así, y así por cada carpeta final (lo que serian las hojas del arbol) donde hayan ficheros. Vamos, que 2 nuevas lineas por hoja de árbol. Bastante lioso, pero la idea es que hay que crear (replicar) la estructura completa del origen en el destino. 
Por lo menos es el método que conozco yo. Si fuese una estructura de cientos de ramificaciones, se pensaría en algo más sofisticado de bucles cmake.

rsantos88, 2016-12-21:
Gracias. Funcionó bastante bien, quedando el fichero resultante ha quedado de la [siguiente forma](https://github.com/rsantos88/teo-head/blob/spanishLanguageModel/share/speechRecognition/CMakeLists.txt).
Los ficheros se instalan correctamente en la ruta deseada. Ahora el siguiente problema a resolver es que YARP encuentre correctamente los ficheros en los subdirectorios que hemos creado a través de su resourceFinder...
```bash
/usr/lib/python2.7/dist-packages/gobject/constants.py:24: Warning: g_boxed_type_register_static: assertion 'g_type_from_name (name) == 0' failed
  import gobject._gobject
Using pygtkcompat and Gst from gi
||| clearing context
||| adding context [speechRecognition]
||| finding file follow-me-english.lm
||| checking [/home/aion/repos/rsantos88/teo-head/programs/speechRecognition/follow-me-english.lm] (pwd)
||| checking [/home/aion/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/follow-me/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-head/robots/default] (robot yarp.d)
||| checking [/home/aion/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/follow-me/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition] (context yarp.d)
||| found /usr/local/share/teo-head/contexts/speechRecognition
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/follow-me-english.lm] (context)
||| checking [/home/aion/.config/yarp/follow-me-english.lm] (YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/follow-me-english.lm] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/follow-me-english.lm] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/follow-me-english.lm] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/follow-me-english.lm] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/follow-me-english.lm] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/follow-me-english.lm] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/follow-me-english.lm] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/follow-me-english.lm] (YARP_DATA_DIRS)
||| checking [/usr/local/share/follow-me/follow-me-english.lm] (yarp.d)
||| checking [/usr/local/share/teo-head/follow-me-english.lm] (yarp.d)
||| did not find follow-me-english.lm
||| finding file follow-me-english.dic
||| checking [/home/aion/repos/rsantos88/teo-head/programs/speechRecognition/follow-me-english.dic] (pwd)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/follow-me-english.dic] (context)
||| checking [/home/aion/.config/yarp/follow-me-english.dic] (YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/follow-me-english.dic] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/follow-me-english.dic] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/follow-me-english.dic] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/follow-me-english.dic] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/follow-me-english.dic] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/follow-me-english.dic] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/follow-me-english.dic] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/follow-me-english.dic] (YARP_DATA_DIRS)
||| checking [/usr/local/share/follow-me/follow-me-english.dic] (yarp.d)
||| checking [/usr/local/share/teo-head/follow-me-english.dic] (yarp.d)
||| did not find follow-me-english.dic
```

jgvictores, 2016-12-23:
1. En https://github.com/rsantos88/teo-head/blob/c01e737e8c6b5d23438e376753415fab8c18f215/share/speechRecognition/CMakeLists.txt#L23 sería `en-us` en lugar de `es-us`, ¿no?
2. Sería incluir las rutas, e.g. `follow-me-english.lm` -> `dictionary/follow-me/follow-me-english.lm` [aquí](https://github.com/rsantos88/teo-head/blob/c01e737e8c6b5d23438e376753415fab8c18f215/programs/speechRecognition/speechRecognition.py#L102) y en lugares parecidos, ¿no?

rsantos88, 2017-01-18:
Esta issue está concluida. Falta por documentar las instrucciones para ampliar las palabras que queramos reconocer. Para el inglés es sencillo,  haciendo uso de esta [herramienta](http://www.speech.cs.cmu.edu/tools/lmtool-new.html). Para el español, hay que seguir una serie de pasos que sería interesante documentar. @jgvictores, ¿dónde lo podríamos poner?

jgvictores, 2017-01-18:
Veo dos opciones, si quieres puedes empezar por la primera:
1. Crear fichero `teo-head-development-manual.md` en la carpeta `doc` y enlazarlo desde la sección Contributing del `README.md`.
1. Crear fichero `teo-head-development-manual.md` en un repositorio nuevo `teo-development-manual` y enlazarlo desde la sección Contributing del `README.md`.

rsantos88, 2017-01-24:
Añadido y enlazado el manual (e0b14e5e8a596daec225de2b5e89ceca3e8e3808).
Cierro issue!!

closed 2017-01-24 by rsantos88