roboticslab-uc3m/vision #34: [FollowMe] Improving demo

rsantos88, 2016-06-28:
Propongo la posibilidad de mejorar la demo de "Follow me" (puesto que actualmente es la única que está operativa) añadiendo algo más credibilidad y vistosidad para cuando alguien quiera ver a Teo. Por ejemplo, se pueden intentar añadir cosas "sencillas" como acompañar el balanceo permanente de brazos por la posibilidad de que al principio el robot mueva el brazo (gesticulando) a la vez que dice "ok, I am following you".

jgvictores, 2016-06-28:
@rsantos88 Could you please post this as an issue in the [follow-me repository](https://github.com/roboticslab-uc3m/follow-me) and close it here?

Thanks!!

rsantos88, 2016-06-28:
Ok @jgvictores. Sorry :)

closed 2016-06-28 by rsantos88