roboticslab-uc3m/vision #38: Create yarp device that wraps espeak

jgvictores, 2016-10-13:
As addressed in https://github.com/roboticslab-uc3m/teo-main/issues/55 the pipe solution of https://github.com/roboticslab-uc3m/teo-main/blob/e0e47c2fc6b3cc40fceffefc075ae811ec0b9b99/share/teoBase/scripts/teoBase.xml#L46-L56 is relatively evil (will not be managed correctly by yarpmanager, and will not be closed correctly as there are two processes and not one).
Create a wrapper of [espeak](http://espeak.sourceforge.net/) similar to https://github.com/robotology/speech/tree/master/svox-speech.

rsantos88, 2016-10-14:
After following the steps, when I try to do "cmake.." in ../espeakExample/build, it says:
  
> By not providing "FindTEO_HEAD.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TEO_HEAD",
  but CMake did not find one.

> Could not find a package configuration file provided by "TEO_HEAD" with any
  of the following names:
    TEO_HEADConfig.cmake
    teo_head-config.cmake

 > Add the installation prefix of "TEO_HEAD" to CMAKE_PREFIX_PATH or set
  "TEO_HEAD_DIR" to a directory containing one of the above files.  If
  "TEO_HEAD" provides a separate development package or SDK, be sure it has
  been installed.

I have opened the .bashrc script and It shows the last line with the path:
> export TEO_HEAD_DIR=/repos/rsantos88/teo-head/build

What can I do?

jgvictores, 2016-10-14:
Please paste the output of:
```bash
tail -5 ~/.bashrc
echo $TEO_HEAD_DIR
ls $TEO_HEAD_DIR
```

jgvictores, 2016-10-14:
Wait just a sec. `TEO_HEAD_DIR` needs the full path. Pre-pend /home/YOUR_USER_NAME/ to it.

rsantos88, 2016-10-14:
I just tried to put the full path and it works ok. Thanks @jgvictores

jgvictores, 2016-10-14:
:smiley:

jgvictores, 2016-10-18:
Lots done at a2be7f8a63b8c285a70ee174bdb0135b9a23cb98.
Missing:
- [ ] Example app
- [ ] Update follow-me
- [ ] Update waiter?

closed 2016-11-11 by rsantos88