roboticslab-uc3m/vision #42: [speechRecognition.py] Installation packages needed to run it correctly

rsantos88, 2016-10-28:
Actualizar documentación para poder ejecutar correctamente [speechRecognition.py] (https://github.com/roboticslab-uc3m/teo-head/blob/develop/programs/speechRecognition/speechRecognition.py) (paquetes, librerias o programas necesarios a instalar para su correcta ejecución y pasos a seguir)

jgvictores, 2016-10-28:
I'd need a clean distro to know,  but I'd at least expect:
```bash
sudo apt-get install python-pocketsphinx
```
We also need the YARP python bindings (this could be added to the YARP section of teo-head) :
```bash
sudo apt-get install swig  # creates bindings
sudo apt-get install libpython-dev # may need python headers
cd yarp/build
ccmake ..
# set YARP_COMPILE_BINDINGS
# press c
# set CREATE_PYTHON or similar 
# press g
make
sudo make install
```

Then, there's lots of gstreamer stuff we need, a.k.a. gstreamerVERSION-plugins-good or similar and also pygst for python. Thing is, there's version 0.1 working on the teo-head PC, and there's 1.0 supported by most recent tutorials: http://cmusphinx.sourceforge.net/wiki/gstreamer
I'd vote for updating the gst.parse_launch like the tutorial's and go for 1.0 (on a branch).

jgvictores, 2016-10-28:
I've left some thoughts above,  but expect this is going to be an iterative process. Don't be afraid of posting long error logs!

rsantos88, 2016-10-28:
Ok, voy a ir documentando los errores que voy encontrando:
```
||| clearing context
||| adding context [speechRecognition]
||| finding file words-20160617.lm
||| checking [/home/aion/repos/repositorio_base/teo-head/programs/speechRecognition/words-20160617.lm] (pwd)
||| checking [/home/aion/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/teo-head/robots/default] (robot yarp.d)
||| checking [/home/aion/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition] (context yarp.d)
||| found /usr/local/share/teo-head/contexts/speechRecognition
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20160617.lm] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20160617.lm
||| finding file words-20160617.dic
||| checking [/home/aion/repos/repositorio_base/teo-head/programs/speechRecognition/words-20160617.dic] (pwd)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20160617.dic] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20160617.dic
yarp: Port /speechRecognition:o active at tcp://192.168.1.42:10002
Traceback (most recent call last):
  File "speechRecognition.py", line 86, in <module>
    app = SpeechRecognition1()
  File "speechRecognition.py", line 41, in __init__
    self.init_gst()
  File "speechRecognition.py", line 47, in init_gst
    + '! pocketsphinx name=asr ! fakesink')
glib.GError: no hay un elemento «gconfaudiosrc»
```

rsantos88, 2016-10-28:
Solución a lo anterior:
> sudo apt-get install gstreamer0.10-plugins-good-dbg

jgvictores, 2016-10-28:
1. `dbg` es versión debug... Antes de que cerremos la issue, se podría reemplazar por un paquete no debug.
2. Ten en cuenta que vas por el camino 0.1, no 1.0 :smiley:

rsantos88, 2016-11-01:
Antes de portar speechRecognition.py a la nueva versión gstreamer 1.0, hay que hacer una correcta instalación de Pocketsphinx y Sphinxbase, la cual se documenta en [Using PocketSphinx with GStreamer and Python](http://cmusphinx.sourceforge.net/wiki/tutorialpocketsphinx) y en [Building application with pocketsphinx] (http://cmusphinx.sourceforge.net/wiki/gstreamer#configuring_the_decoder_and_improving_accuracy)

He sintetizado los pasos a seguir:
```
sudo apt-get install libgstreamer1.0-dev
sudo apt-get install libgstreamer-plugins-base1.0-dev

# Installation sphinxbase
cd
cd repos
git clone https://github.com/cmusphinx/sphinxbase.git
cd sphinxbase
./autogen.sh
./configure
make -j4
make check # to be sure that all is ok
sudo make install

# Installation pocketsphinx
cd
cd repos
git clone https://github.com/cmusphinx/pocketsphinx
cd pocketsphinx
./autogen.sh
./configure
make -j4 clean all
make -j4 check
sudo make install
 
# We need to configure the path to look for shared libaries
echo "export LD_LIBRARY_PATH=/usr/local/lib" >> ~/.bashrc
echo "export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig" >> ~/.bashrc
echo "export GST_PLUGIN_PATH=/usr/local/lib/gstreamer-1.0" >> ~/.bashrc

# To verify that GStreamer can find the plugin. You should get a large amount of output, ending with something like this:
# decoder             : The underlying decoder
#                        flags: readable
#                        Boxed pointer of type "PSDecoder"

gst-inspect-1.0 pocketsphinx

# to test that all is runnin ok
sudo modprobe snd_pcm_oss
pocketsphinx_continuous -inmic yes
```

rsantos88, 2016-11-01:
Tras intentar migrar la versión se speechRecognition.py a la nueva versión de GStreamer 1.0 (3f8303beb5c77abe98eeb1a78ed89049f583b44d),  salta el siguiente error:
```
Using pygtkcompat and Gst from gi
||| clearing context
||| adding context [speechRecognition]
||| finding file words-20150720.lm
||| checking [/home/aion/repos/rsantos88/teo-head/programs/speechRecognition/words-20150720.lm] (pwd)
||| checking [/home/aion/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/teo-head/robots/default] (robot yarp.d)
||| checking [/home/aion/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/aion/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition] (context yarp.d)
||| found /usr/local/share/teo-head/contexts/speechRecognition
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.lm] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20150720.lm
||| finding file words-20150720.dic
||| checking [/home/aion/repos/rsantos88/teo-head/programs/speechRecognition/words-20150720.dic] (pwd)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic
yarp: Port /speechRecognition:o active at tcp://192.168.1.42:10002
Traceback (most recent call last):
  File "speechRecognition.py", line 97, in <module>
    app = SpeechRecognition()
  File "speechRecognition.py", line 48, in __init__
    self.init_gst()
  File "speechRecognition.py", line 60, in init_gst
    asr.connect('result', self.asr_result)
TypeError: <__main__.GstPocketSphinx object at 0xb423c2fc (GstPocketSphinx at 0xa29c920)>: unknown signal name: result
```

jgvictores, 2016-11-01:
From the [Code listing](http://cmusphinx.sourceforge.net/wiki/gstreamer#code_listing) I can see that things have changed. Instead of:
```python
asr = self.pipeline.get_by_name('asr')
asr.connect('result', self.asr_result)
```
It should be more like:
```python
bus = self.pipeline.get_bus()
bus.add_signal_watch()
bus.connect('message::element', self.element_message)
```
and then interpret and do whatever required like they do in the connected `element_message` function:
```python
def element_message(self, bus, msg):
```
PS: Maybe we could still keep the `asr = self.pipeline.get_by_name('asr')` if it's still useful for setting the dictionary.

rsantos88, 2016-11-02:
Migración realizada de gstreamer 0.10 a 1.0 (#43) en speechRecognition.py
Tras integrar los cambios para el cambio de diccionario (ver [aquí](https://github.com/rsantos88/teo-head/blob/issue-32-asr/programs/speechRecognition/speechRecognition.py)) y pasarle por el puerto rpc "setDictionary test", el programa rompe con el siguiente mensaje:
```
Using pygtkcompat and Gst from gi
||| clearing context
||| adding context [speechRecognition]
||| finding file words-20150720.lm
||| checking [/home/teo/repos/rsantos88/teo-head/programs/speechRecognition/words-20150720.lm] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/follow-me/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-body/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-head/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo/robots/default] (robot yarp.d)
||| checking [/home/teo/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/follow-me/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-body/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition] (context yarp.d)
||| found /usr/local/share/teo-head/contexts/speechRecognition
||| checking [/usr/local/share/teo/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.lm] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20150720.lm
||| finding file words-20150720.dic
||| checking [/home/teo/repos/rsantos88/teo-head/programs/speechRecognition/words-20150720.dic] (pwd)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic
swig/python detected a memory leak of type 'Type *', no destructor found.
yarp: Port /speechRecognition:o active at tcp://2.2.2.118:10036
yarp: Port /speechRecognition/rpc:s active at tcp://2.2.2.118:10037
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2,000000e+00
-allphone				
-allphone_ci		yes		yes
-alpha			0.97		9,700000e-01
-ascale			20.0		2,000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1,000000e-20
-bestpath		yes		yes
-bestpathlw		9.5		9,500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic
-dictcase		no		no
-dither			no		no
-doublebw		no		no
[... 78 lines cut ...]
-unit_area		yes		yes
-upperf			6855.4976	6,800000e+03
-uw			1.0		1,000000e+00
-vad_postspeech		50		50
-vad_prespeech		20		20
-vad_startspeech	10		10
-vad_threshold		3.0		3,000000e+00
-var					
-varfloor		0.0001		1,000000e-04
-varnorm		no		no
-verbose		no		no
-warp_params				
-warp_type		inverse_linear	inverse_linear
-wbeam			7e-29		7,000000e-29
-wip			0.65		6,500000e-01
-wlen			0.025625	2,562500e-02

in DataProcessor.read
Trying to read from connection
Received [setDictionary test]
Running test... You can say: Hi, food, people, exit
terminate called after throwing an instance of 'Swig::DirectorMethodException'
Abortado (`core' generado)

```

rsantos88, 2016-11-02:
El problema se ha solucionado eventualmente con el uso de self en las respectivas variables (45a0273ef6ee812b5f4b049186dfc18f9153f707).
En cambio, a la hora de cambiar el diccionario, tira el siguiente error:
```
Using pygtkcompat and Gst from gi
||| clearing context
||| adding context [speechRecognition]
||| finding file words-20150720.lm
||| checking [/home/teo/repos/rsantos88/teo-head/programs/speechRecognition/words-20150720.lm] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/follow-me/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-body/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo-head/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo/robots/default] (robot yarp.d)
||| checking [/home/teo/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/follow-me/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-body/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition] (context yarp.d)
||| found /usr/local/share/teo-head/contexts/speechRecognition
||| checking [/usr/local/share/teo/contexts/speechRecognition] (context yarp.d)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.lm] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20150720.lm
||| finding file words-20150720.dic
||| checking [/home/teo/repos/rsantos88/teo-head/programs/speechRecognition/words-20150720.dic] (pwd)
||| checking [/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic] (context)
||| found /usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic
swig/python detected a memory leak of type 'Type *', no destructor found.
yarp: Port /speechRecognition:o active at tcp://2.2.2.118:10036
yarp: Port /speechRecognition/rpc:s active at tcp://2.2.2.118:10037
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2,000000e+00
-allphone				
-allphone_ci		yes		yes
-alpha			0.97		9,700000e-01
-ascale			20.0		2,000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1,000000e-20
-bestpath		yes		yes
-bestpathlw		9.5		9,500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/share/teo-head/contexts/speechRecognition/words-20150720.dic
-dictcase		no		no
-dither			no		no
-doublebw		no		no
[... 129 lines cut ...]

(python:7483): GStreamer-CRITICAL **: 
Trying to dispose element audioconvert0, but it is in PLAYING instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

(python:7483): GStreamer-CRITICAL **: 
Trying to dispose element autoaudiosrc0, but it is in PLAYING instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

(python:7483): GStreamer-CRITICAL **: 
Trying to dispose element pipeline0, but it is in PAUSED instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

terminate called after throwing an instance of 'Swig::DirectorMethodException'
Abortado (`core' generado)
```

rsantos88, 2016-11-03:
Tema resuelto (9d907d563aa7d5d090348042f574987dd02c1263).
 :tada: :tada: :tada: :tada:

rsantos88, 2016-11-03:
Se ha realizado la actualización de la [documentación](https://github.com/roboticslab-uc3m/teo-head/blame/develop/doc/teo_head_install_on_ubuntu_15_04.md#L80-L124) para la correcta ejecución de speechRecognition.py con la nueva versión de gstreamer 1.0.
Se da por concluida esta issue :+1:

closed 2016-11-03 by rsantos88