roboticslab-uc3m/vision #46: [speechRecognition.py] Error in Yarp Module Manager

rsantos88, 2016-11-04:
Actualmente, con la nueva versión de speechRecognition.py instalada en teo-head, el Yarp Module Manager no es capaz de realizar el arranque remoto del mismo, mostrando el siguiente error:
` [ERR]:   (teoBase_App) cannot run /sbin/runuser on /head : cannot run /sbin/runuser on /head`

rsantos88, 2016-11-04:
Pongo esta información por si pudiera ser de utilidad:
Al ejecutar en terminal los pasos que realizaría el xml del Yarp Manager:
```bash
ssh head
cd /sbin
sudo runuser -l teo -c speechRecognition.py
```
Nos muestra el siguiente error:
```bash 
Using pygtkcompat and Gst from gi
gdk_mir_display_open
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Unable to init server: Could not connect: Connection refused
gdk_mir_display_open
Failed to connect to Mir: Failed to connect to server socket: No such file or directory
Unable to init server: Could not connect: Connection refused
||| clearing context
||| adding context [speechRecognition]
||| finding file words-20150720.lm
||| checking [/home/teo/words-20150720.lm] (pwd)
||| checking [/home/teo/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/home/teo/repos/teo-head/build/share/teo-head/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/speech/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/teo/repos/teo-head/build/share/teo-head/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/speech/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/teo/.config/yarp/contexts/speechRecognition] (context YARP_CONFIG_HOME)
||| checking [/home/teo/.local/share/yarp/contexts/speechRecognition] (context YARP_DATA_HOME)
||| checking [/etc/yarp/contexts/speechRecognition] (context YARP_CONFIG_DIRS)
||| checking [/home/teo/repos/teo-head/build/share/teo-head/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| found /home/teo/repos/teo-head/build/share/teo-head/contexts/speechRecognition
||| checking [/usr/local/share/speech/contexts/speechRecognition] (context YARP_DATA_DIRS)
||| checking [/home/teo/repos/teo-head/build/share/teo-head/contexts/speechRecognition/words-20150720.lm] (context)
||| found /home/teo/repos/teo-head/build/share/teo-head/contexts/speechRecognition/words-20150720.lm
||| finding file words-20150720.dic
||| checking [/home/teo/words-20150720.dic] (pwd)
||| checking [/home/teo/repos/teo-head/build/share/teo-head/contexts/speechRecognition/words-20150720.dic] (context)
||| found /home/teo/repos/teo-head/build/share/teo-head/contexts/speechRecognition/words-20150720.dic
swig/python detected a memory leak of type 'Type *', no destructor found.
yarp: Port /speechRecognition:o active at tcp://2.2.2.53:10009
yarp: Port /speechRecognition/rpc:s active at tcp://2.2.2.53:10010
Traceback (most recent call last):
  File "/usr/local/bin/speechRecognition.py", line 180, in <module>
    app = SpeechRecognition()
  File "/usr/local/bin/speechRecognition.py", line 108, in __init__
    self.init_gst()
  File "/usr/local/bin/speechRecognition.py", line 119, in init_gst
    + '! pocketsphinx name=asr beam=1e-20 ! fakesink')
GLib.Error: gst_parse_error: no element "pocketsphinx" (1)
```

jgvictores, 2016-11-05:
This link looks helpful: http://www.devsplanet.com/question/35931585
In case link ever dies, I'd highlight 'gst-inspect-1.0 pocketsphinx' (through `runuser` in our case), and the use of the environment variable `GST_PLUGIN_PATH`.

rsantos88, 2016-11-07:
Al ejecutar por terminal la linea `sudo runuser -l teo -c "gst-inspect-1.0 pocketsphinx"` obtenemos el siguiente resultado:
```bash
/sbin$ sudo runuser -l teo -c "gst-inspect-1.0 pocketsphinx"
No such element or plugin 'pocketsphinx'
```
Este error era consecuencia de que no estaba especificada la ruta de gstreamer en `GST_PLUGIN_PATH`.
Una vez pasada la ruta al script `.bashrc`, el problema se solucionaba (ver [aqui](https://github.com/roboticslab-uc3m/teo-head/commit/6be6f63d309cde88a7d463750c4a17e4891186c9#diff-20d433e2502ae00262c6cda32cc34fb3R112))
Es posible que runuser este ignorando este script al ejecutarse...

rsantos88, 2016-11-07:
Tema resuelto (ver [commit](https://github.com/roboticslab-uc3m/teo-main/commit/bab6d27b522a8ecb61f3adc4752e2bae7f65dc4c) de teo-main)
Cierro esta issue :)

closed 2016-11-07 by rsantos88

jgvictores, 2018-05-17:
This issue was moved to roboticslab-uc3m/speech#26