roboticslab-uc3m/vision #52: [vision] Think about ViSP

jgvictores, 2016-11-17:
[vision] Think about ViSP: http://visp.inria.fr

PeterBowman, 2018-10-27:
We have some info related to visual servoing scattered around, see:

* https://github.com/roboticslab-uc3m/vision/issues/52
* https://github.com/roboticslab-uc3m/legacy-matlab-visual-servo
* https://github.com/roboticslab-uc3m/follow-me/issues/34#issuecomment-433612540

jgvictores, 2020-01-05:
> [vision] Think about ViSP: http://visp.inria.fr

Added at https://github.com/roboticslab-uc3m/vision/commit/2f97fa81d80980cc8aa19f5eebf9a009c3b06ada

> We have some info related to visual servoing scattered around, see:
> 
> * #74
> * https://github.com/roboticslab-uc3m/legacy-matlab-visual-servo
> * [roboticslab-uc3m/follow-me#34 (comment)](https://github.com/roboticslab-uc3m/follow-me/issues/34#issuecomment-433612540)

Added at https://github.com/roboticslab-uc3m/vision/commit/e45834fb3404e6ad8d0b8ae3bdd2c4cbec1cd907

closed 2020-01-05 by jgvictores