roboticslab-uc3m/vision #60: Imminent repo name change (teo-head -> vision)

jgvictores, 2017-03-02:
On Friday March 3rd 2017 evening, this repo will be renamed: teo-head -> vision

Related with https://github.com/roboticslab-uc3m/QA/issues/2 and https://github.com/orgs/roboticslab-uc3m/projects/2

As a consequence, the Espeak device will be moved to roboticslab-uc3m/yarp-devices (see https://github.com/roboticslab-uc3m/teo-body/issues/99 )

Cheers!

PeterBowman, 2017-03-04:
Renaming done. All references to the old location should redirect now to `vision`, including fetch/push operations. In order to update your remotes, see [Changing a remote's URL](https://help.github.com/articles/changing-a-remote-s-url/). I'd leave this issue open until every single reference in the organization has been updated accordingly: [search](https://github.com/search?q=org%3Aroboticslab-uc3m+teo-head&type=Code).

PeterBowman, 2017-03-05:
I've tracked down and replaced all (but one) occurrences of `teo-head` in the repos at our org. I didn't know how to proceed with this:

* [follow-me/doc/fig/editable/follow-me-app.graphml#L421](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c/doc/fig/editable/follow-me-app.graphml#L421)

jgvictores, 2017-03-06:
[follow-me/doc/fig/editable/follow-me-app.graphml](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c/doc/fig/editable/follow-me-app.graphml) is a [yEd](https://www.yworks.com/products/yed)-editable file from which [follow-me/doc/fig/follow-me-app.png](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c3b61c999836419587aef5d1d5586dd2bd/doc/fig/follow-me-app.png) is generated. We are actually going to have to make quite a few updates on it.

jgvictores, 2017-04-18:
Removed speech already at https://github.com/roboticslab-uc3m/speech at 680e9dd21e0cd83b0c6fa5da5a4bba6cdc725a2e.

jgvictores, 2017-04-22:
1. I've started a new issue https://github.com/roboticslab-uc3m/follow-me/issues/33 regarding [comment](https://github.com/roboticslab-uc3m/vision/issues/60#issuecomment-284267998) by @PeterBowman :
> I've tracked down and replaced all (but one) occurrences of `teo-head` in the repos at our org. I didn't know how to proceed with this: [follow-me/doc/fig/editable/follow-me-app.graphml#L421](https://github.com/roboticslab-uc3m/follow-me/blob/c1ba54c/doc/fig/editable/follow-me-app.graphml#L421)

2. Also related with the transition, I'll be working on #64: Remove big speech files from history

Closing and will work on individual issues.

closed 2017-04-22 by jgvictores