roboticslab-uc3m/vision #64: Remove big speech files from history

jgvictores, 2017-04-22:
As described in #60 and issues commented there, this repo is now only vision. I will proceed to try to strip this repo from old speech files (now in https://github.com/roboticslab-uc3m/speech) using https://rtyley.github.io/bfg-repo-cleaner/

Once done, it will be recommended to freshly clone this repo to avoid re-uploading the purged big files.

jgvictores, 2017-04-22:
```bash
git clone --mirror git://github.com/roboticslab-uc3m/vision.git
java -jar ../../src/bfg-1.12.15.jar --delete-files '{*.lm,*.dic,*.dict,mdef,means,mixture_weights,sendump,shm_humanoid_controller,variances}' vision.git
```
Results in deleting (biggest was `es-20k.lm` at 66.5 MB):
```
b79913f6ebca940bd0d2c5986e0f8a16efe5d807 601223 es-20k.dic
93eed47d91859b4dad5999e6535bc23b9861646d 69689826 es-20k.lm
b79913f6ebca940bd0d2c5986e0f8a16efe5d807 601223 es.dic
b79913f6ebca940bd0d2c5986e0f8a16efe5d807 601223 es.dict
883ad48d0e490f13d8ddc0d380bfa18bbd7de249 65 follow-me-20150720.dic
947fdea8e0cd5735d0e0df67417646c894cf12a0 990 follow-me-20150720.lm
883ad48d0e490f13d8ddc0d380bfa18bbd7de249 65 follow-me-english.dic
d133ac2347763b0cdb8f644f86b33b5ac6fb75db 114 follow-me-english.dic
3ed8c2fad5ef8fdcb757e0765357f47a2f6fddac 93 follow-me-english.dic
947fdea8e0cd5735d0e0df67417646c894cf12a0 990 follow-me-english.lm
b7e2d86e1d5aa516e0717ece9c958867db08cdd7 1202 follow-me-english.lm
ba0a918084a64dc741dc5e6ca126eb4b31eb7a30 1351 follow-me-english.lm
9d38f061c6b0c7491f80bf604815acfbef2f53e2 66 follow-me-spanish.dic
1542c0470bbdf806ab98e9222b0ae13bdc006993 76 follow-me-spanish.dic
9372bdd3e288a3904e5ada36e0db258d86af2081 466 follow-me-spanish.lm
0c49a8987076b156307694465c67df09c795eef6 401 follow-me-spanish.lm
8a5e8ee48bb01643d7a38bf67f09046fe93f3230 1351979 mdef
597f72c19652451268deee0b84554d96be50fa34 6992020 mdef
0959f8fe43b7e06e21074525842cf672708ab032 23620676 means
89f8108cdbb15386b399b33c343cba31c99a71f8 519244 means
027dbbe62e1e384df2c45fe3e057a1b19c8f48b3 6263872 mixture_weights
30311cd51ba444534afd008d440c7cdcb534290e 656192 mixture_weights
61ccd61e842ada2326c0d7b830dee31b9a3b8fb4 1566592 sendump
3210494e17f74b20396ea76017aae79cc51f6d83 883951 shm_humanoid_controller
5fbe3c7eb64ceb90b4d2bd539324ba78164a1990 136 test-spanish.dic
ee617dee45af64b0a1bb677bf274f05874db36df 1399 test-spanish.lm
e50e611a09b2295d78d1be7d399834e34c4c1bd2 77 testSpeech.dic
9cf1666ecfe62309950b1a65c7b361cbfb663654 1013 testSpeech.lm
6fb1f968ff6a0bd9ac21ab14f784442bbe24c13e 519244 variances
6aaacdc4f129790bc5e50b31e04366630be62677 23620676 variances
cac1c186ff666dc0fbcc14d42175a80301df2145 59 waiter-20161103.dic
50e510113d9d73a95ce5748760d405911ac4c6ea 1017 waiter-20161103.lm
4b2f84c5496bcf455ecea5de82bbb32efdb54f85 113 waiter-english.dic
2674f30d53f73196da7fd46c778601d4d727d4d1 1349 waiter-english.lm
72d57327f69adf156aff2ea097072d9faf527756 142 waiter-spanish.dic
964dfcff4c17a817899bd9b29b6a3e0b99351117 108 waiter-spanish.dic
d8ffc24cf6295de7df572e04e54809c60df936dd 1380 waiter-spanish.lm
8bf1e45a6be5abeca8eb1b1c9f23ad3ec1a945ce 554 waiter-spanish.lm
883ad48d0e490f13d8ddc0d380bfa18bbd7de249 65 words-20150720.dic
947fdea8e0cd5735d0e0df67417646c894cf12a0 990 words-20150720.lm
f78c9fd1e9daf4bd256b1a5a19168557b2ec3baa 990 words-20150720.lm
cf34c34cede0a85c6b15902cd5309097fac0694e 154 words-20160617.dic
9c3e0707e98b44bb475fa7ca150c130710a360dc 1505 words-20160617.lm
```

jgvictores, 2017-04-22:
Getting feedback like:
```
! [remote rejected] refs/pull/17/head -> refs/pull/17/head (deny updating a hidden ref)
```
Which are apparently harmless. However, a downloaded repo is apparently always 136.8M and has not gone down...

jgvictores, 2017-04-22:
Same with:
```bash
java -jar ../../src/bfg-1.12.15.jar --strip-blobs-bigger-than 1M vision.git
```
(which would at least catch the biggest unwanted files).

jgvictores, 2017-04-22:
Was able to do the per-file filtering, yay!

The key was in ([ref](https://rtyley.github.io/bfg-repo-cleaner/)):

> The BFG treats you like a reformed alcoholic: you've made some mistakes in the past, but now you've cleaned up your act. Thus the BFG assumes that your latest commit is a good one, with none of the dirty files you want removing from your history still in it.

So removed remaining files from `share` first, at 5bbaac6b7f0e449f3abb7ded2c4a97b3c292a37c, merged into `master` at f5e832bf5b49fc19a9fdd753564cfe466d9244c8. Done!

jgvictores, 2017-04-22:
PS: Please freshly clone this repo to avoid re-uploading the purged big files!!!

jgvictores, 2019-07-29:
Seems like these files are present. Reopening for revision.

PeterBowman, 2019-07-29:
It doesn't look like history has been ever rewritten in this repo, or at least not with the BFG tool. Commits previous to april, 22th (2017) do not contain hashes that do show up in other repos with rewritten commits, e.g. kinematics-dynamics (see https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/fbb09f1065be580295a3514cf5ad3b27bad06efb).

PeterBowman, 2021-04-25:
> Commits previous to april, 22th (2017) do not contain hashes that do show up in other repos with rewritten commits, e.g. kinematics-dynamics (see [roboticslab-uc3m/kinematics-dynamics@fbb09f1](https://github.com/roboticslab-uc3m/kinematics-dynamics/commit/fbb09f1065be580295a3514cf5ad3b27bad06efb)).

Well, perhaps the hash mess was just an accident, I'm not 100% sure if this tool is supposed to always add that information.

> Seems like these files are present. Reopening for revision.

Have we actually tracked these leftover big files? I can't find them in the repo tree at the time you wrote that comment ([sources](https://github.com/roboticslab-uc3m/vision/tree/379cd305d66fcd1234022ced3a10daec6218d2a3)). I've just cloned this repo and it weighs 16 MB, which doesn't seem like a pretty big deal. It's almost 7 MB only on [doc/fig](https://github.com/roboticslab-uc3m/vision/tree/c53efe22ba3ea279a555c0ad6b08f4463c981eba/doc/fig) and [share/HaarDetector/conf](https://github.com/roboticslab-uc3m/vision/tree/c53efe22ba3ea279a555c0ad6b08f4463c981eba/share/HaarDetector/conf) combined, and 7.6 MB on the .git directory, which could point at potentially heavy files still lingering somewhere. Yet it's still not that bad. On the other hand, rewriting a repo's history is prone to cause havoc, ping users if ever mentioned in a commit, similarly many issues might be referenced again...

@jgvictores could we close this issue, then?

jgvictores, 2021-04-25:
Sure, thank you so much!

closed 2021-04-25 by PeterBowman