roboticslab-uc3m/vision #67: Integrate travislib into this repo

PeterBowman, 2017-05-03:
Several code bits are already included in the `colorRegionDetection` program. Since travislib has a rather tiny code base, let's create a separate library here and eliminate that repo to avoid duplication and simplify builds.

jgvictores, 2017-05-03:
I think we can keep the [travislib repository](https://github.com/roboticslab-uc3m/travislib) for historical reasons and code such as [optical flow, etc](https://github.com/roboticslab-uc3m/travislib/blob/master/contrib/superseded/travisOriginal/travis.cpp).

Regarding the copies of `TravisLib.cpp` and `TravisLib.hpp` in this repo, perhaps they could be moved to `libraries` to promote code reuse.

PeterBowman, 2017-07-14:
I guess that improvements on TravisLib would end up here (vision), and the travislib repo will remain mostly as it is.

PeterBowman, 2018-01-18:
> Since travislib has a rather tiny code base, let's create a separate library here (...)

Done long ago at 3be1606.

> I think we can keep the travislib repository for historical reasons and code such as optical flow, etc.

OK!

closed 2018-01-18 by PeterBowman

jgvictores, 2018-01-19:
Wow! Hadn't noticed that! Thanks a lot!