roboticslab-uc3m/vision #68: Think about grip

jgvictores, 2017-05-05:
Think about grip: http://wpilib.screenstepslive.com/s/4485/m/24194/l/463566-introduction-to-grip

jgvictores, 2017-05-05:
Java code https://github.com/WPIRoboticsProjects/GRIP

jgvictores, 2017-10-25:
Related: https://github.com/asrob-uc3m/air-hockey/issues/5

jgvictores, 2018-12-12:
Add description in README at 7086718. Closing due to inactivity.

David-Estevez, 2018-12-12:
I did try GRIP for a few things (ASROB and Horus)... do you have any doubts that I might be able to solve? It's a pity when a poor issue is closed due to inactivity :sweat_smile:

jgvictores, 2018-12-12:
All is fine. I've also used GRIP for a few personal projects, and it's great!

I'm just trying to close all these eternal "think about" issues. I'll document this new "think about" workflow at https://github.com/roboticslab-uc3m/developer-manual once it's more stable! :smile:

It's referenced at https://github.com/roboticslab-uc3m/vision/blob/develop/README.md#similar-and-related-projects Please edit if you think more description is required, and thanks for the offer and expression of interest!

closed 2022-06-04 by PeterBowman