roboticslab-uc3m/vision #69: Create common API for vision modules with client/server device pair

PeterBowman, 2017-07-15:
Old [ColorClient library](https://github.com/roboticslab-uc3m/asibot-main/tree/c975cd5b6e049be9a59e1c11b807faefe51fee6f/src/libraries/ColorClientLib) from asibot-main repo might be a start, see also roboticslab-uc3m/asibot-main#28.

jgvictores, 2020-11-01:
Fists API has been added at https://github.com/roboticslab-uc3m/vision/commit/7497b4b07688aafa101889f5285abddc1c16c427 which was oriented at solving https://github.com/roboticslab-uc3m/vision/issues/79

jgvictores, 2020-11-01:
Re-opening to treat the client/server device pair

PeterBowman, 2020-12-24:
This issue relates to exposing `IDetector::detect` in an ICP manner (RPC-driven):

https://github.com/roboticslab-uc3m/vision/blob/3ed7898f8b57d45f1725b945e863fe4e6a2b6ec0/libraries/YarpPlugins/IDetector.hpp#L22-L23

Let's assume we have a remote server accepting input frames and returning a vector of detected objects. The client part encodes this data in a similar manner (sends YARP frames, expects YARP vectors). Client code (which instantiates the client device) is responsible for frame acquisition from a camera, be it local or remote (but ideally it should be remote).

Now consider this: in the above scenario 1. the client application connects to the remote camera, 2. the frame travels through YARP network and arrives to the app, 3. the app sends this frame again to the server device by means of the client counterpart, 4. the server processes the request and prepares a response, 5. the response arrives to the client app through the client device.

Note how the image frame must make two stops before arriving to the expected destination. Sure, it could undergo some processing on the client app itself (e.g. drawing the ROI of the detected objects), but still it feels wrong that the server device does not ask for the frame on a regular basis. We can work this around by connecting the server to a remote camera so that any incoming request initiates the detection on frames acquired on demand. However, this would introduce an assymetry between using the underlying detector subdevice locally (must be fed with a frame) or remotely (the server knows how to request a frame from an external source).

So, as spoken with @jgvictores, we expect the server-client pair to address a specific scenario: simulation. The client app fetches frames from a simulated environment (locally and at full framerate to reduce latency?). The server resides on a different process, maybe even on a different CPU better suited for heavy computations if that's the case.

@jgvictores please correct me if I got this wrong.

jgvictores, 2020-12-28:
@PeterBowman Yes, this is the main idea. I think it is the cleanest implementation.

Note that I am open to alternatives regarding the possibility of receiving and processing a stream. However, it would require adding logic to the device, which we want to avoid. In fact, the logic of the obvious use case is already conveniently covered in https://github.com/roboticslab-uc3m/vision/tree/master/programs/rgbDetection ([perma](https://github.com/roboticslab-uc3m/vision/tree/3ed7898f8b57d45f1725b945e863fe4e6a2b6ec0/programs/rgbDetection)). We should also recall that there is a 2nd use case, that uses different interfaces, covered in https://github.com/roboticslab-uc3m/vision/tree/master/programs/rgbdDetection ([perma](https://github.com/roboticslab-uc3m/vision/tree/3ed7898f8b57d45f1725b945e863fe4e6a2b6ec0/programs/rgbdDetection)) which is in fact more used in practice (https://github.com/roboticslab-uc3m/follow-me).

PeterBowman, 2022-01-14:
I think this could make sense:

* Define our new API in [vision.thrift](https://github.com/roboticslab-uc3m/vision/blob/ad9c273e688b3aab86fa755fff1ebcddd85d73da/libraries/VisionIDL/vision.thrift).
* Make [*rgbDetection*](https://github.com/roboticslab-uc3m/vision/tree/ad9c273e688b3aab86fa755fff1ebcddd85d73da/programs/rgbDetection) and [*rgbdDetection*](https://github.com/roboticslab-uc3m/vision/tree/ad9c273e688b3aab86fa755fff1ebcddd85d73da/programs/rgbdDetection) act as servers that implement said interface.
* Both apps use `IDetector::detect` internally, therefore they are responsible for acquiring sensor frames and passing them to the locally instantiated detector device. In other words, frames only travel once (unless the sensor is also instantiated locally), processing is also local to *rgb(d)Detection*.
* I'd rename the "/state:o" ports to something like "/detector:o" and use them solely for debugging purposes, i.e. the output of `detect` shall be broadcast as-is, also in the depth case.
* The server implementation of the new IDL API will perform all extra processing necessary. In the case of *rgbdDetection*, this entails extracting the XYZ coordinates.
* Prepare examples and bindings.

Now, since the paradigm will switch from a stream-based communication to RPC, I think the new API should reflect the latest successful detection and its timestamp. Client apps will decide whether to discard stale results.

I'm still unsure regarding the data structure: should it be a `Bottle`, a list of `Property` containers, a map or a list of maps? In the RGBD scenario we are only interested in a single detected object (the closest one) and XYZ coordinates, whereas in the RGB case it's all of them in frame coords. This disparity can justify the introduction of two methods instead of one, let's say: `getLastPreferredDetection()` and `getLastDetections()`.

PeterBowman, 2022-01-14:
Proposal: `return_rgb getLastRgbDetection()` where:

```thrift
struct return_rgb {
    1: bool ret = false;
    2: i32 tlx;
    3: i32 tly;
    4: i32 brx;
    5: i32 bry;
    6: double timestamp;
}
```

To be implemented by *rgbDetection* and *rgbdDetection*. Aditionally, this service could feature start/stop detection commands and frame size getters. Also, `tlx`/`tly` and `brx`/`bry` could live in their own `Pixel` structure.

The, only *rgbdDetection* would implement `return_depth getLastDepthDetection()`, where:

```thrift
struct return_depth {
    1: bool ret = false;
    2: double x
    3: double y;
    4: double z;
    5: double timestamp;
}
```

I'm thinking of two more methods to complement the previous ones, While these only return one detection according to the best predefined criteria (closest to the frame center or nearest Z), a sibling method would return all available simultaneous detections without such distinction.

jgvictores, 2022-01-14:
In **both** cases (rgb + rgbd), I'm inclined to **always** return some sort of array. This was our original rgb approach, where we had, e.g. the maxBlob or similar parameter. More generally, depending on **the algorithm itself** or **the algorithm config parameters**, the actual elements in the array can be limited to 1.

PeterBowman, 2022-01-17:
ASWJ, we are not going to implement this considering the current state of things. The IDL proposal is somewhat nicer than a new pair of YARP devices in the usual client/server arrangement, albeit not as much worthy on a second thought. Sadly, the YARP-IDL implementation differs from the ROS message system in that we can mimic their RPC services, but broadcasting data through topics is not available in the YARP world (to the best of my knowledge). In this vein, it doesn't feel great to force back-and-forth data flow in the RPC pull model (client sends requests, server provides reply) as compared to the current streaming solution. In the former scenario, communication is initialized on the client side (ideally those commands would originate from a periodic thread), whereas the latter entails a simple port callback as usual. We concluded that switching to a well-defined C++ protocol is not going to pose a clear advantage given the use cases we know of.

A few interesting ideas arised from this, though. It would be nice to have a configuration mechanism in *rgb(d)Detection*. An RPC port would allow this on runtime, but we can also cope with having this ability on initialization only. The discussed options all boil down to how the "best" detection is selected, which could be one of these:

1. closest to the frame center
2. closest to the sensor Z-wise
3. largest area
4. ...others?

This filter may apply to one or several elements. For instance, a `--maxNumBlobs N` option would limit the detection to the `N` largest objects only.

In addition, maybe *rgbdDetection* could publish not only XYZ, but also pixel coordinates. Implementing this on the same streaming port is a violation of the protocol, though.

Finally, as noted by @jgvictores, YARP veers toward (ab)using of port monitors. There is one prominent example: a port monitor capable of performing image segmentation. It can be attached either to the sender or the receiver, thus the choice of the place where (potentially) heavy computations occur would be left to the programmer. Since our use cases allow local execution, a port-only solution as this may be considered suboptimal.

jgvictores, 2022-01-18:
> In addition, maybe _rgbdDetection_ could publish not only XYZ, but also pixel coordinates. Implementing this on the same streaming port is a violation of the protocol, though.

Yes, totally. `rgbDetection` already does this. You can see how the information the actual detector passes in a bottle https://github.com/roboticslab-uc3m/vision/blob/ad9c273e688b3aab86fa755fff1ebcddd85d73da/programs/rgbDetection/RgbDetection.cpp#L158-L160 is directly published at https://github.com/roboticslab-uc3m/vision/blob/ad9c273e688b3aab86fa755fff1ebcddd85d73da/programs/rgbDetection/RgbDetection.cpp#L199-L200

This is the idea of an array of dictionaries I've been consistent with. I believe this would be the best option for _rgbdDetection_ too (I understand it breaks protocol with e.g. teo-follow-me, but that side can/should be updated).

PD: I'm be writing more comments up next.

PeterBowman, 2022-01-18:
> I believe this would be the best option for _rgbdDetection_ too...

Exactly, in this proposal I wanted said port to stream raw, unmodified detection data, i.e. a dictionary (represented as a YARP bottle) which stores pixel coordinates and may vary slightly across different detectors. In contrast, the extra RPC port would accomodate its protocol to the command of choice (one object vs several objects, pixel coordinates vs XYZ coordinates).

> ...(I understand it breaks protocol with e.g. teo-follow-me, but that side can/should be updated).

However, I think it is **not** feasible to make *rgbdDetection* publish such data in the current state of things. It is *rgbdDetection*'s responsibility to convert pixel coordinates to XYZ coordinates. To achieve this, it needs to query the intrinsic parameters from the sensor and apply a known formula. I wouldn't like to place this burden on teo-follow-me or other client applications.

jgvictores, 2022-01-18:
ASWB, to sum up, there were many interesting ideas in the conversation, and I'd also like to write them down for posterity. There are some consensus that I'm just writing here (again) explicitly:

1. We're cool with the `IDetector` as a local plugin/device. The plugin/device mechanism enables a clean interface, with selection of the implementation at runtime.
1. We're cool with the current architecture, where `rgbDetector`/`rgbdDetector`:
   1. Loads a sensor device.
   1. Loads a `IDetector` plugin/device.
   1. Thus acts as glue between (i) and (ii).
   1. Publishes the processed data. Bonus points: It would be nice to update the port name to `/detected:o`.

While this architecture may look like it implies the sensor and detector must live on the same machine (remember, we said the detector is local), this is not true: decoupling via port can occur on the sensor side, as the _sensor_ device can be **local** _or_ **remote**.

More stuff!

- Regarding statefullness,  the _detector_ could technically have, regarding states: none, constant (1 during entire lifetime), or varying (potentially several during entire lifetime) states. Being:
    - No states: This is **not** the current case nor what we want. Most detectors are passed some configuration parameters upon initialization, thus providing some sort of notion of state.
    - Constant state (1 during entire lifetime): This is the **current situation**. It works. Most detectors are passed some configuration parameters upon initialization, which defines the state, and additionally remains constant throughout all the lifetime of the detector. We can speak about their actual role and maybe a cleanup/standardization, but I'll touch this topic later on in this same comment.
    - Varying state (potentially several during entire lifetime): This is doable, but IMHO **not required** ATOW. For completeness, I would like to mention the 2 steps required for an implementation of this within the current architecture: a) Expose a _detector_ method for re-configuration. b) Have `rgbDetector`/`rgbdDetector` expose an RPC port (could be implemented via IDL) which forwards reconfiguration to the detector. Note that this can lead to additional confusion if some RPC commands are for the detector and some are to stay at the `rgbDetector`/`rgbdDetector` level.

Now, regarding actual parameters (note: **which** and **where**!):
- It is pretty clear at times (e.g. visual-servoing) we only want a **single** object, selected following some specific criteria.
- For now, it's easier and okay to define the criteria upon initialization, see above.
- So, there are 2 (very related) issues left to solve:
    - Is the "filter"/"selection" performed on the _detector_ side, the `rgbDetector`/`rgbdDetector` wrapper, or both (which seems uglier but difficult to avoid)? What I mean is, `rgbDetector`/`rgbdDetector` seems like a good place, but we should take into account that some detectors already have parameters that involve some kind of object filter/selection (e.g. YOLO's threshold on IOU/Jaccard; potentially some minSize in colorRegion...).
    - Best naming scheme? Recently I've thought of the use of best with superlatives: `--best closest`, `--best biggest` `--best centered` which also supports opposites (cannot find the use case, but just for completion): `--best furthest`, `--best smallest`. Note that in a classic exercise of seeking to be extremely flexible, I've also though about the pair: `--sort area` `--maxObjects 1` or similar.

jgvictores, 2022-01-18:
> I wouldn't like to place this burden on teo-follow-me or other client applications.

I don't know... at first though, I'd have no issue with a client performing 2 simple operations of arithmetic mean over 2 elements.

PeterBowman, 2022-01-18:
> I don't know... at first though, I'd have no issue with a client performing 2 simple operations of arithmetic mean over 2 elements.

I would certainly have an issue if all clients needed to open their own RGBD client device, fetch a depth frame, perform RGB/depth alignment, query the intrinsic parameters and, finally, apply the pin-hole model. That's indeed a lot of burden to place on clients and again raises some of my concerns from https://github.com/roboticslab-uc3m/vision/issues/69#issuecomment-750909956, but in a worse manner: two processes need to access a sensor, and at least one of them forcibly does that in a remote manner.

In other words: I don't want teo-follow-me (and other apps) to deal with the sensor. I only need the result of the detection. If that's not enough, then the streaming/debug port would provide raw data to work with for my specific needs.

> Is the "filter"/"selection" performed on the _detector_ side, the `rgbDetector`/`rgbdDetector` wrapper, or both (which seems uglier but difficult to avoid)

The wrapper seems a good place for centralization of efforts: why implement the same set of filters on each detector if it can be done once in the caller (*rgb(d)Detection*)? Some of those parameters may not apply to all wrapped detectors, but this is subject to proper documentation and should not be a problem, even if a detector can be initialized in a way that involves some kind of inherent filtering as the one you mentioned.

jgvictores, 2022-01-18:
> I would certainly have an issue if all clients needed to open their own RGBD client device, fetch a depth frame, perform RGB/depth alignment, query the intrinsic parameters and, finally, apply the pin-hole model.

I really hadn't taken into account this portion of logic that was happening on the `rgbdDetector` side. I totally agree. : )

closed 2022-06-04 by PeterBowman