roboticslab-uc3m/vision #70: RGBDsensor ROS support still not implemented

Kuronogard, 2017-10-06:
related: https://github.com/roboticslab-uc3m/tiago-main/issues/9#issuecomment-334727290

davidvelascogarcia, 2018-04-16:
https://github.com/robotology/yarp/blob/b806be8c6c3c6e67190e8ec1e428cf3e9e1c2b64/src/libYARP_dev/src/devices/RGBDSensorClient/RGBDSensorClient.cpp

PeterBowman, 2018-04-16:
Server support is already there ([ref](https://github.com/robotology/yarp/blob/b45be5e9c36f3b3b75a57492a4e4982a5fa762fc/src/libYARP_dev/src/devices/RGBDSensorWrapper/RGBDSensorWrapper.cpp#L480-L506)), perhaps the client implementation would not be that hard. Adding *upstream* label.

jgvictores, 2018-04-16:
Yes, ee are thinking of a PR at some point. :+1:

davidvelascogarcia, 2018-04-20:
https://github.com/roboticslab-uc3m/vision/compare/ROSInputTopics

PeterBowman, 2018-04-20:
> https://github.com/roboticslab-uc3m/vision/compare/ROSInputTopics

Hm, just to keep track of what is going on here: does this task refer to missing ROS support in RGBDSensorClient (as I assumed based on previous comments, hence the label), or is it more focused on **local** devices/apps, such as Dennis's work on *haarDetection*?

@davidvelascogarcia there are several auto-generated files in your current branch that do not belong to version control, namely `*.pro`, `*.pro.user` (added to .gitignore at https://github.com/roboticslab-uc3m/vision/commit/135413e5bf32c05a3101bb33e9757c8390bb66e4), and mostly anything in the `msg/` directory since it can be generated by `yarpidl_rosmsg` and CMake targets.

Kuronogard, 2018-04-20:
> at 135413e

This is about missing ROS support in RGBDSensorClient.

My work on haarDetection and colorRegionDetection was actually a 
workaround to receive images directly from ROS topics, not from local 
devices. The idea was to deprecate them in favour of RGBDSensorClient 
when they implement the ROS support (or implement it ourselves).

Sorry if it was not clear enough.

davidvelascogarcia, 2018-04-23:
@PeterBowman You´re right. It´s temporal while i´m trying to develop the code. I´m doing tests and using  the branch, git and github like a backup sorry.

PeterBowman, 2018-04-23:
No problem, just noting that this branch would require some serious history rewriting prior to any merge into develop. Anyway, since this seems to be a project of yours, have you considered working with a fork?

jgvictores, 2018-05-31:
Priority lowered and unassigned, focus will be on:
- https://github.com/orgs/roboticslab-uc3m/projects/7?card_filter_query=label%3A%22priority%3A+high%22
- https://github.com/orgs/roboticslab-uc3m/projects/7?card_filter_query=label%3A%22status%3A+in+progress%22

PeterBowman, 2020-08-25:
WIP at https://github.com/robotology/yarp/pull/2346.

PeterBowman, 2020-09-14:
So, https://github.com/robotology/yarp/pull/2346 has been merged into current `master` (YARP 3.5.0, unreleased). It adds a new `RGBDFromRosTopic` device: a YARP-ROS bridge aimed for being wrapped by `RGBDSSensorWrapper` (similarly to existing acquisition devices, e.g. `realsense2` or `depthCamera`). In turn, this means that `RGBDSensorClient` can be used to interface remotely just like with any other YARP-enabled camera. If I'm correct, there is no need to change our apps in any way.

The outdated (2017) `ROSInputTopics` branch (https://github.com/roboticslab-uc3m/vision/compare/1ed72d0...1b1e3a7) has been rendered pointless.

PeterBowman, 2020-09-14:
> The outdated (2017) `ROSInputTopics` branch ([1ed72d0...1b1e3a7](https://github.com/roboticslab-uc3m/vision/compare/1ed72d0...1b1e3a7)) has been rendered pointless, I guess.

Same as `issue-70-RGBDsensor-ROS-support-still-not-implemented` (https://github.com/roboticslab-uc3m/vision/compare/1de27a1...c4637b3)?

PeterBowman, 2020-09-14:
I can confirm the ROS-YARP-RGBD bridge works, it seems even less CPU-consuming. See https://github.com/IntelRealSense/realsense-ros for install instructions (despite the suggestions, I have installed the ROS package over an existing Intel SDK). There might be some kind of issue with the `depthimage(2)` carriers (PR is coming, I presume), but at least depth frames are arriving at destination. Both branches have been deleted, commits should be preserved via above XXX...YYY commit links though.

Check http://www.yarp.it/classRGBDSensorFromRosTopic.html.

closed 2020-09-14 by PeterBowman

jgvictores, 2020-09-14:
@PeterBowman This is good news! Thank you so much for handling this!

jgvictores, 2020-11-11:
> I can confirm the ROS-YARP-RGBD bridge works, it seems even less CPU-consuming. See https://github.com/IntelRealSense/realsense-ros for install instructions (despite the suggestions, I have [...]

Also confirmed for Gazebo simulation of TIAGo! `yarpdev --device RGBDSensorWrapper --subdevice RGBDSensorFromRosTopic --rgb_data_topic /xtion/rgb/image_color --rgb_info_topic /xtion/rgb/camera_info --depth_data_topic /xtion/depth_registered/image_raw --depth_data_info /xtion/depth_registered/camera_info --name /jtest`