roboticslab-uc3m/vision #74: Use yarp::dev::IVisualServoing in device implementations

PeterBowman, 2017-12-17:
Scheduled for the YARP v2.3.72 release, this is the header file at current `devel`: [IVisualServoing.h](https://github.com/robotology/yarp/blob/6ad3d9c19b527f9d87450b41b3f8e5941a0b614d/src/libYARP_dev/include/yarp/dev/IVisualServoing.h).

Up to now, we've developed these visual servoing apps:

* [followMeHeadExecution@follow-me](https://github.com/roboticslab-uc3m/follow-me/tree/75d7590e449c16cee060f40334d54d29e286f439/programs/followMeHeadExecution) + [haarDetection](https://github.com/roboticslab-uc3m/vision/tree/eaa267b9b73100418f879ab909fc564a36d4af67/programs/haarDetection)
* [haarDetectionController@kinematics-dynamics](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/7ead8123613356f46b05e10a477c2fbf014f920f/programs/haarDetectionController) + [haarDetection2D](https://github.com/roboticslab-uc3m/vision/tree/eaa267b9b73100418f879ab909fc564a36d4af67/programs/haarDetection2D)
* [matlab/visualServo@asibot-main](https://github.com/roboticslab-uc3m/asibot-main/tree/2171d3e6a1ce16ac92f29a939b9dc85db7245dda/matlab/visualServo)
* ...

jgvictores, 2017-12-17:
That's interesting, hadn't seen that development!

PeterBowman, 2018-06-07:
Upstream docs: http://www.yarp.it/classyarp_1_1dev_1_1IVisualServoing.html.

PeterBowman, 2018-10-27:
See also:

* https://github.com/roboticslab-uc3m/legacy-matlab-visual-servo
* https://github.com/roboticslab-uc3m/follow-me/issues/34#issuecomment-433612540

jgvictores, 2019-05-09:
Added [robotology/visual-tracking-control](https://github.com/robotology/visual-tracking-control) (https://robotology.github.io/visual-tracking-control) at new section in README at https://github.com/roboticslab-uc3m/vision/commit/f2bb1e9001738061ea2ff2f12a4a1b9a68149f5a.

Interestingly, https://robotology.github.io/visual-tracking-control/doxygen/doc/html/modules.html has `hand-tracking`, `visualservoingclient`, `visualservoingserver`.

PeterBowman, 2021-04-30:
Closing due to inactivity and low priority/lack of interest. Moreover, this YARP interface contains many more methods than we actually need, and it seems oriented towards dual camera configurations. I should add that our [haarDetectionController](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/9b476c96c1af2aea975703615123bee66d14fdcd/programs/haarDetectionController) app is mostly abandoned, although should still work. Perhaps it should be moved to a demo repo or such because of its ad-hoc nature. By the way, we also developed a "centroid transform" module for the [streamingDeviceController](https://github.com/roboticslab-uc3m/kinematics-dynamics/tree/9b476c96c1af2aea975703615123bee66d14fdcd/programs/streamingDeviceController) app, achieving some sort of negotiated control in teleoperation scenarios. None of these use cases justify the need for a specialized interface implementation.

closed 2022-06-04 by PeterBowman