roboticslab-uc3m/vision #77: Work on haar

jgvictores, 2018-01-30:
In private repo issue https://github.com/roboticslab-uc3m/tiago-main/issues/9 there were attempts to train haar:
> [Here](https://www.dropbox.com/sh/0gi5gpekmybuin1/AADNr1-ilpV_vjUj_IkveyORa?dl=0) you have the positive and negative samples, the .vec file for _opencv_haarcascade_ and the command lines I used for for the different generation/training steps (only as reference, as the paths are invalid)

The objective of this issue is to track down what went wrong, and develop documentation of what made it work if it turns out to be feasible via haar.

davidvelascogarcia, 2018-02-19:
Note: our haar `xml` files are currently stored at https://github.com/roboticslab-uc3m/vision/tree/develop/share/haarDetection/conf

davidvelascogarcia, 2018-02-19:
WIP at https://github.com/roboticslab-uc3m/vision/tree/issue-77-haar/share/haarDetection/conf

jgvictores, 2018-11-07:
Entiendo que esto es un duplicado de #77. @davidvelascogarcia ¿Ok cerrar issue y eliminar rama, seguro que no se nos pierde ningún fichero?

davidvelascogarcia, 2018-11-07:
Sí, #77 es un duplicado de la branch [tiago-vision](https://github.com/roboticslab-uc3m/vision/tree/tiago-vision) del [pull request](https://github.com/roboticslab-uc3m/vision/pull/90). Se pueden borrar sin problema y cerrar la branch [issue-77](https://github.com/roboticslab-uc3m/vision/tree/issue-77-haar) e issue #77. Todos los cambios que se realizaron en [issue-77](https://github.com/roboticslab-uc3m/vision/tree/issue-77-haar), han sido pasados a  [tiago-vision](https://github.com/roboticslab-uc3m/vision/tree/tiago-vision). Acabo de revisarlo otra vez por si acaso, solo faltaba añadir una referencia al `README` que ya ha sido añadida. e6096a31aeaecbd3e2287ba5e910717c6072925f

jgvictores, 2018-11-07:
Cerrando!

closed 2022-06-04 by PeterBowman