roboticslab-uc3m/vision #79: Merge programs (no hurry, just a reminder)

jgvictores, 2018-02-09:
- colorRegionDetection / colorRegionDetection2D
- haarDetection / haarDetection2D

...and perhaps all together. :-)

davidvelascogarcia, 2019-11-11:
Regarding #79 issue i´ve been trying to merge `colorRegionDetection` /` colorRegionDetection2D` and `haarDetection` / `haarDetection2D`.
First, i´ve started trying to merge `colorRegionDetection2D` with `haarDetection2D`. 
Updates in [vision](https://github.com/davidvelascogarcia/vision/tree/haarColorRegionDetection2D) fork , `haarColorRegionDetection2D` branch. Right now works correctly at the same time and publish the results in two different ports `haar` and `colorRegion` ports.  
I have to check features ports, it is added but i haven´t test yet.
Picture test attached.

[image]
Figure 1. Test haar and color region detection

Regarding the name, i didn´t know the name to be selected and now i have named as `haarColorRegionDetection2D`, as fusion´s name.
@jgvictores what do you thing about that?

PeterBowman, 2019-11-11:
Is it worth in terms of efficiency having this app perform two different tasks at once?

jgvictores, 2019-11-11:
Nope! @davidvelascogarcia and I have spoken regarding the proposed architecture. It would look more like the following:

[image]

PeterBowman, 2019-11-11:
Looks nice. I'd keep those libs embedded within the haarColorRegionDetection2D application, unless you're really sure this could be highly reusable (given the heavy YARP constraints, I'm not sure about it). I presume that switching between modes is going to be done on init via YARP option?

jgvictores, 2019-11-11:
I was not thinking in keeping the `haarColorRegionDetection2D` name. My original idea intended to be able to switch to tensorflow too.

Yes, I guess the config will involve passing a YARP option. The API will involve passing a YARP image and getting the features in return.

davidvelascogarcia, 2019-11-25:
I have been working with proposed architecture in [switchDetection2D](https://github.com/davidvelascogarcia/vision/tree/switchDetection2D) branch. 
The name that i´ve selected provisionally is `switchDetection2D`.

`switchDetection2D` module, merge vision `detection2D` programs. The module merge `colorRegionDetection2D`, `haarDetection2D` and `tensorflowDetection2D`. 
This module works with `segmentorThread`. Using vision `detection2D` programs structure. 
Build an object of selected detection mode, the mode process the image received  and return the processed image and results, to be published.
PR at [#78](https://github.com/roboticslab-uc3m/vision/pull/98).

The process diagram attached in Figure 1.
[image]
Figure 1. Diagram

Now i´m going to add [Transformation](https://github.com/roboticslab-uc3m/tools/blob/develop/libraries/YarpPlugins/RealToSimControlboard/Transformation.hpp) class to build the selected detector objects using parameters objects instead of each parameter individually.

jgvictores, 2020-11-01:
Edited description because it was way too ambitious. Enough has been done at https://github.com/roboticslab-uc3m/vision/commit/7497b4b07688aafa101889f5285abddc1c16c427 (see commit message for more tasks that were accomplished in the process):
- Old: colorRegionDetection2D + haarDetection2D
- New: switchDetection2D + libs (HaarDetector + ColorRegionDetector)

closed 2020-11-01 by jgvictores