roboticslab-uc3m/vision #81: fail installation OpenCV3 on 16.04

jmgarciah, 2018-02-20:
when you want to install on 16.04, it is not detected.

$ sudo apt-get install libtiff4-dev

could it be a possible solution the lib "libtiff5-dev" ¿?¿?

PeterBowman, 2018-02-20:
Our guides ([ref](https://github.com/roboticslab-uc3m/installation-guides/blob/349a48be4b9768089901eeb97d85264e239c123f/install-opencv.md#install-opencv-3-with-contrib-and-python-3-support)) state that `libtiff4-dev` is needed, but, as you pointed out, there is no such package on Xenial. On the other hand, `libtiff5-dev` is available from Trusty onwards:

* https://packages.ubuntu.com/search?keywords=libtiff4-dev
* https://packages.ubuntu.com/search?keywords=libtiff5-dev

Official guides recommend installing `libtiff-dev`, which redirects to `libtiff5-dev` on those systems:

* https://docs.opencv.org/3.0-beta/doc/tutorials/introduction/linux_install/linux_install.html
* https://packages.ubuntu.com/search?keywords=libtiff-dev

jgvictores, 2018-02-20:
@jmgarciah Hi! Thanks for pointing this out!

> could it be a possible solution the lib "libtiff5-dev" ¿?¿?

@jmgarciah Yes, that is a possible solution!

> Official guides recommend installing libtiff-dev, which redirects to libtiff5-dev on those systems:

@PeterBowman Great finding, thanks! I've updated at https://github.com/roboticslab-uc3m/installation-guides/commit/57cd9733c80312a262fd8b8d7b71fd10a95b999a to provide this more portable solution.

@jmgarciah Feel free to close this issue if you got things working on the libtiff side!

jmgarciah, 2018-02-21:
thx to @PeterBowman and @jgvictores

closed 2018-02-21 by jmgarciah