roboticslab-uc3m/vision #82: Error compilation  - bolbsXY

jmgarciah, 2018-02-22:
```
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp: In member function ‘virtual void roboticslab::SegmentorThread::run()’:
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:117:5: error: ‘vector’ was not declared in this scope
     vector<cv::Point2d> blobsXY;
     ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:117:5: note: suggested alternative:
In file included from /usr/include/c++/5/vector:64:0,
                 from /usr/include/c++/5/bits/random.h:34,
                 from /usr/include/c++/5/random:49,
                 from /usr/include/c++/5/bits/stl_algo.h:66,
                 from /usr/include/c++/5/algorithm:62,
                 from /usr/local/include/opencv2/hal/defs.h:570,
                 from /usr/local/include/opencv2/core/cvdef.h:59,
                 from /usr/local/include/opencv2/core/types_c.h:59,
                 from /usr/local/include/opencv2/core/core_c.h:48,
                 from /usr/local/include/opencv/cv.h:63,
                 from /home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.hpp:15,
                 from /home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:3:
/usr/include/c++/5/bits/stl_vector.h:214:11: note:   ‘std::vector’
     class vector : protected _Vector_base<_Tp, _Alloc>
           ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:117:23: error: expected primary-expression before ‘>’ token
     vector<cv::Point2d> blobsXY;
                       ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:117:25: error: ‘blobsXY’ was not declared in this scope
     vector<cv::Point2d> blobsXY;
                         ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:123:12: error: expected primary-expression before ‘double’
     vector<double> blobsAngle,blobsArea,blobsAspectRatio,blobsAxisFirst,blobsAxisSecond,blobsPerimeter;
            ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:124:12: error: expected primary-expression before ‘double’
     vector<double> blobsRectangularity,blobsSolidity;
            ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:125:12: error: expected primary-expression before ‘double’
     vector<double> blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev;
            ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:126:25: error: ‘blobsArea’ was not declared in this scope
     travis.getBlobsArea(blobsArea);
                         ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:127:30: error: ‘blobsPerimeter’ was not declared in this scope
     travis.getBlobsPerimeter(blobsPerimeter);
                              ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:128:29: error: ‘blobsSolidity’ was not declared in this scope
     travis.getBlobsSolidity(blobsSolidity);
                             ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:129:24: error: ‘blobsHue’ was not declared in this scope
     travis.getBlobsHSV(blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev);
                        ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:129:33: error: ‘blobsSat’ was not declared in this scope
     travis.getBlobsHSV(blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev);
                                 ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:129:42: error: ‘blobsVal’ was not declared in this scope
     travis.getBlobsHSV(blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev);
                                          ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:129:51: error: ‘blobsHueStdDev’ was not declared in this scope
     travis.getBlobsHSV(blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev);
                                                   ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:129:66: error: ‘blobsSatStdDev’ was not declared in this scope
     travis.getBlobsHSV(blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev);
                                                                  ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:129:81: error: ‘blobsValStdDev’ was not declared in this scope
     travis.getBlobsHSV(blobsHue,blobsSat,blobsVal,blobsHueStdDev,blobsSatStdDev,blobsValStdDev);
                                                                                 ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:130:34: error: ‘blobsAngle’ was not declared in this scope
     if( ! travis.getBlobsAngle(0,blobsAngle) )  // method: 0=box, 1=ellipse; note check for return as 1 can break
                                  ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:135:32: error: ‘blobsAspectRatio’ was not declared in this scope
     travis.getBlobsAspectRatio(blobsAspectRatio,blobsAxisFirst,blobsAxisSecond);  // must be called after getBlobsAngle!!!!
                                ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:135:49: error: ‘blobsAxisFirst’ was not declared in this scope
     travis.getBlobsAspectRatio(blobsAspectRatio,blobsAxisFirst,blobsAxisSecond);  // must be called after getBlobsAngle!!!!
                                                 ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:135:64: error: ‘blobsAxisSecond’ was not declared in this scope
     travis.getBlobsAspectRatio(blobsAspectRatio,blobsAxisFirst,blobsAxisSecond);  // must be called after getBlobsAngle!!!!
                                                                ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:136:35: error: ‘blobsRectangularity’ was not declared in this scope
     travis.getBlobsRectangularity(blobsRectangularity);  // must be called after getBlobsAngle!!!!
                                   ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:176:34: error: ‘blobsAngle’ was not declared in this scope
                 output.addDouble(blobsAngle[0]);
                                  ^
/home/teo/repos/vision/programs/colorRegionDetection2D/SegmentorThread.cpp:179:37: error: ‘blobsAngle’ was not declared in this scope
                 for (int i = 0; i < blobsAngle.size(); i++)
                                     ^
programs/colorRegionDetection2D/CMakeFiles/colorRegionDetection2D.dir/build.make:86: fallo en las instrucciones para el objetivo 'programs/colorRegionDetection2D/CMakeFiles/colorRegionDetection2D.dir/SegmentorThread.cpp.o'
make[2]: *** [programs/colorRegionDetection2D/CMakeFiles/colorRegionDetection2D.dir/SegmentorThread.cpp.o] Error 1
CMakeFiles/Makefile2:359: fallo en las instrucciones para el objetivo 'programs/colorRegionDetection2D/CMakeFiles/colorRegionDetection2D.dir/all'
make[1]: *** [programs/colorRegionDetection2D/CMakeFiles/colorRegionDetection2D.dir/all] Error 2
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error `2```
```

jmgarciah, 2018-02-22:
sorry, the namespaces 'cv' and 'std' are not include.

If you add in SegmentorThread.cpp, the problem is solved.

PeterBowman, 2018-02-22:
Yeah, but this is a mess anyway... I'm going to apply https://github.com/roboticslab-uc3m/developer-manual/issues/19 here and review namespaces, too.

PeterBowman, 2018-02-22:
@jmgarciah namespace directive added at dbd4916 and merged into develop, please check and close the issue if solved. I'll keep looking for unresolved or non-explicit #include's.

PeterBowman, 2018-02-26:
> sorry, the namespaces 'cv' and 'std' are not include.
> 
> If you add in SegmentorThread.cpp, the problem is solved.

OK, I guess it's done. Closing.

closed 2018-02-26 by PeterBowman