roboticslab-uc3m/vision #83: Drop old OpenNI2 interfaces, adopt IRGBDSensor

PeterBowman, 2018-02-24:
YARP devel has discontinued support for several devices, affecting Kinect/OpenNI2 devices currently used in this repo. This means that the `yarp/dev/IOpenNI2DeviceDriver.h` header (as well as `yarp/dev/IKinectDeviceDriver.h`, by the way) is no longer installed: [YARP v3.0.0 Release Notes (unreleased)](https://github.com/robotology/yarp/blob/3136ec4854cda8426827f5b4f39bfd1eaad65d5d/doc/release/v3_0_0.md#libraries).

There is a replacement, though. Client apps must switch to the [`yarp::dev::RGBDSensorClient`](http://www.yarp.it/classyarp_1_1dev_1_1RGBDSensorClient.html) device and the [`yarp::dev::IRGBDSensor`](http://www.yarp.it/classyarp_1_1dev_1_1IRGBDSensor.html) interface, which should be barely invasive on our end (the vision repo). Real camera devices are now wrapped by [`yarp::dev::depthCameraDriver`](http://www.yarp.it/classyarp_1_1dev_1_1depthCameraDriver.html) (goes by the .ini name *depthCamera*).

jgvictores, 2018-02-25:
Yes, this is definitively something we'll have to cope with. No libs for now, just a couple of programs. Transition should not be difficult.

PeterBowman, 2018-03-20:
Raising priority. According to https://github.com/robotology/yarp/projects/6, we should solve this before 2018-05-12.

PeterBowman, 2018-04-11:
WIP at 0099fd1, testing on Kinect v1 and v2, compiles at latest YARP devel ([CI build](https://travis-ci.org/roboticslab-uc3m/vision/builds/365216998)). Problem: YARP throws a `Trying to allocate an invalid pixel type image` and makes the program quit ([ref](https://github.com/robotology/yarp/blob/245690a2bb2b371a81efe0da2f9dbbd9f34e53b2/src/libYARP_sig/src/Image.cpp#L338-L342)) upon calling `IRGBDSensor::getImages`.

jgvictores, 2018-04-11:
I'd add a `colorFrame.setPixelCode(VOCAB_PIXEL_RGB);` immediately after https://github.com/roboticslab-uc3m/vision/blob/0099fd11d24a0ee8adf4c95c94d6c6059b9d2b42/programs/colorRegionDetection/SegmentorThread.cpp#L143

jgvictores, 2018-04-11:
((did some similar stuff around https://github.com/roboticslab-uc3m/openrave-yarp-plugins/commit/f72ff821f102cd562242b2864e5a301e231a4094#diff-203b1994e1821fea53f07aea4e9df0eb))

PeterBowman, 2018-04-11:
Did that, didn't help.

PeterBowman, 2018-04-13:
Fixed by applying a small delay prior to starting the segmentor thread: https://github.com/roboticslab-uc3m/vision/commit/1367d0ec1a4d1356b61ed4c7f2d24147296fc051. Unfortunately, [`IRGBDSensor::getSensorStatus`](http://www.yarp.it/classyarp_1_1dev_1_1IRGBDSensor.html#ababe6dc59f59ef9db2f896ce4581030f) is of no help here, it keeps returning [`RGBD_SENSOR_OK_IN_USE`](http://www.yarp.it/classyarp_1_1dev_1_1IRGBDSensor.html#a007f8ea5d2cbcdcac2eaaf117d47eb25) even if the first frames have invalid pixel codes.

This might deserve an upstream issue regarding the [somewhat draconian `std::exit` call](https://github.com/robotology/yarp/blob/245690a2bb2b371a81efe0da2f9dbbd9f34e53b2/src/libYARP_sig/src/Image.cpp#L338-L342), which was introduced back in 2006.

jgvictores, 2018-04-13:
I'd also look into why status is returning OK. It's probably referring to the yarp connection status, but it should definitely check if correctly formed frames are available.

PeterBowman, 2018-04-13:
> It's probably referring to the yarp connection status

Nope, it's strictly bound to the device layer, not to the network wrapper. All it does is check device and stream readiness for publishing data (even if malformed): [ref](http://www.yarp.it/depthCameraDriver_8cpp_source.html#l01066).

jgvictores, 2018-04-14:
Hack 0417b4055d7ae7c4e1db702189955f0f1465ab92 is pretty ugly/dangerous. Could we add an `upstream` label here?

PeterBowman, 2018-04-20:
Addendum: review and update .ini files across robot repos (<robot>-configuration-files).

PeterBowman, 2018-04-20:
Also: review WIP branches (e.g. [ROSInputTopics](https://github.com/roboticslab-uc3m/vision/compare/ROSInputTopics)) per https://github.com/roboticslab-uc3m/vision/issues/70#issuecomment-383098270.

PeterBowman, 2018-05-18:
Note: installation of *depthCamera* is broken at YARP 2.3.72 (and probably at 2.3.70.x, too). Did work at 2.3.68.x, fixed upstream at 2.3.72.1 (unreleased, see https://github.com/robotology/yarp/pull/1633). Workaround: do `cmake . -DYARP_HAS_OpenNI2=ON` and configure again.

PR #86 (2018-10-31): [WIP] Migrate to yarp::dev::IRGBDSensor interface
Closes #83. Notes:

* Minimum required YARP version established at 2.3.68.
* Mind this hack: https://github.com/roboticslab-uc3m/vision/issues/83#issuecomment-381096322.
* Tested on Kinect v1 + *colorRegionDetection* and *haarDetection* apps. While working with a Kinect v2 device, be aware of #85.

---

# TODO (WIP)

- [ ] we no longer create the RGBD device locally, a client device is opened instead (current default: *RGBDSensorClient*) and connects to the corresponding network wrapper -> partially restore previous behavior and allow local dev [... cut]
closed 2018-10-31 by jgvictores