roboticslab-uc3m/vision #85: Depth frames don't scale well on Kinect v2 devices

PeterBowman, 2018-04-13:
While testing #83, I've been playing around a bit with Kinect v1/v2. These are the frame resolutions as reported by the debugger:

| frame | Kinect v1  | Kinect v2 |
| ------------- | ------------- | ------------- |
| RGB  | 640x480  | 1920x1080  |
| depth  | 640x480  | 640x480  |

Sadly, the *colorRegionSegmentor* app is not prepared for working with unbalanced screen resolutions and keeps throwing segfaults at the following instruction:

https://github.com/roboticslab-uc3m/vision/blob/1de27a1ca0d3f7c45840b8b549680d714a8e59e3/programs/colorRegionDetection/SegmentorThread.cpp#L236

Since `blobsXY` originates from the RGB frame, pixel lookups on the depth frame usually fail due to exceeding matrix bounds. More apps could be affected by this bug.

jgvictores, 2018-04-14:
General statements, also informative for #86:
- In the deprecated OpenNI2 device, we had `--printVideoModes` to see both RGB and Depth modes. Modes set resolution and framerate, and may vary among device and even firmware.
- For TEO, we fixed the RGB and Depth mode parameters for the specific sensor+firmware so resolution would match: https://github.com/roboticslab-uc3m/teo-configuration-files/blob/b37b041ddcfddfe1347cfb4d5c22737d0aa069d2/share/teoBase/scripts/teoBase.xml#L36

Options for this specific issue:
1. Just check if resolutions match, if not just error and exit.
2. Scale one of the the inputs to match the other (I'd add a warning, as scaling may introduce a bit of processing overhead).

PeterBowman, 2018-11-02:
See also https://github.com/roboticslab-uc3m/vision/pull/86#issuecomment-381318636.

PeterBowman, 2019-04-22:
Affects:

https://github.com/roboticslab-uc3m/vision/blob/834c33c1f0795dcd536853949a041c84490d29a7/programs/colorRegionDetection/SegmentorThread.cpp#L238

https://github.com/roboticslab-uc3m/vision/blob/834c33c1f0795dcd536853949a041c84490d29a7/programs/haarDetection/SegmentorThread.cpp#L130-L132

And then:

https://github.com/roboticslab-uc3m/vision/blob/834c33c1f0795dcd536853949a041c84490d29a7/programs/kinectPxToReal/CallbackPort.cpp#L30-L37

Instead of direct acess to the RGB frame, this app processes single requested pixels, therefore depth frame bounds should be check. Probably, this port should also expect the RGB frame size as an additional value in the incoming bottles. Either that, or configure on init (`--scaleX` and `--scaleY`?).

PeterBowman, 2019-04-22:
> Either that, or configure on init (`--scaleX` and `--scaleY`?).

A couple of pixels yields no information about the source frame, we are actually specifying a ratio (coordinates in a frame of unknown size) and that's just half the information needed to reconstruct an XY position. I'd rather tamper with the protocol and require another two elements in the incoming bottles, holding the source frame's width and height, respectively.

PeterBowman, 2019-04-25:
Can't test this on a realsense2 device - depth resolution is enforced to match the RGB resolution.

jgvictores, 2019-04-25:
Would a Asus Xtion Pro Live be helpful?

PeterBowman, 2019-05-05:
> Can't test this on a realsense2 device - depth resolution is enforced to match the RGB resolution.

...unless `needAlignment` is set to `false`, see [docs](http://www.yarp.it/classyarp_1_1dev_1_1realsense2Driver.html).

closed 2019-05-05 by PeterBowman