roboticslab-uc3m/vision #92: Develop c++ tensorflow detector module

jgvictores, 2018-12-12:
Develop c++ tensorflowDetection2D module, similar to `colorRegionDetection2D` and `haarDetection2D`. Default to VGG16 using pre-trained weights on ImageNet.

davidvelascogarcia, 2019-01-08:
He estado probando la API en C++, realizando algunos ejemplos de [Tensorflow C++ API](https://www.tensorflow.org/guide/extend/cc), en el repositiorio [Tensorflow CPP Examples](https://github.com/davidvelascogarcia/tensorflow-cpp-examples), pero han dado algunos problemas las librerías. Era capaz de encontrar las librerías, podía declarar correctamente espacio nombres, instanciar objetos de la clase, incluso funcionaba bien el auto-completado de escritura. Sin embargo a la hora de compilar, salta la advertencia de:

`
/usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../include/c++/4.8/bits/c++0x_warning.h:32:2: error: 
      This file requires compiler and library support for the ISO C++ 2011
      standard. This support is currently experimental, and must be enabled with
      the -std=c++11 or -std=gnu++11 compiler options.
`

y adicionalmente:

`
      'unsupported/Eigen/CXX11/Tensor' file not found
#include "unsupported/Eigen/CXX11/Tensor"
`
En cuanto a la compilación he intentado forzarlo en el CMakeLists.txt, pero no da resultado. En cuanto al problema del Eigen, he probado instalándolo y no ha dado resultado.

PeterBowman, 2019-01-08:
Please check [How to activate C++ 11 in CMake? (SO)](https://stackoverflow.com/q/10851247). Anyway, if you read the [C++ guide](https://www.tensorflow.org/guide/extend/cc) carefully:

> The C++ API is only designed to work with TensorFlow `bazel build`.

Moreover:

> `'unsupported/Eigen/CXX11/Tensor' file not found #include "unsupported/Eigen/CXX11/Tensor"`

Have you tried this Bazel thing? From https://github.com/tensorflow/tensorflow/issues/13705#issuecomment-338633212:

> Anyway, using Bazel resolved the errors about Eigen. It seems like it downloads it automatically somehow.

It looks like there is a workaround for this build system of theirs, though, which means we could use CMake after all: https://github.com/PatWie/tensorflow-cmake. You are hardcoding some header paths [in here](https://github.com/davidvelascogarcia/tensorflow-cpp-examples/blob/0ab1f3781b88d144bd0b15a16e438a8ef4a1a980/programs/example-1/CMakeLists.txt#L8), this is not a valid approach. Instead, a `find_package(TensorFlow)` call in your `CMakeLists.txt` should locate the system paths of your TF installation ([example](https://github.com/PatWie/tensorflow-cmake/blob/675970503a1cffbc033d900c4d528d2d6ae73ed6/inference/cc/CMakeLists.txt)).

davidvelascogarcia, 2019-01-09:
@PeterBowman Thanks, i am going to try it. Maybe the problem is that i am not using Bazel. I am using the libraries directly.

jgvictores, 2019-01-13:
I'm not a big fan of the whole `bazel`+`your code has to be inside the Tensorflow repo` deal.

@davidvelascogarcia maybe you could check the [cmake alternative](https://github.com/PatWie/tensorflow-cmake) as commented by @PeterBowman above?

jgvictores, 2019-01-13:
@davidvelascogarcia PS: also see comment above on using `find_package(TensorFlow)` and avoiding hard-coded paths.

jgvictores, 2019-01-14:
PS: I also find the C API quite interesting.

davidvelascogarcia, 2019-01-25:
@jgvictores @PeterBowman Thanks, right now i can find *Tensorflow* libraries with cmake and without using bazel, but i have a problem looking for *libtensorflow_cc.so*

```
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting TensorFlow using python3 (use -DPYTHON_EXECUTABLE=... otherwise)
-- Detecting TensorFlow info - done
-- Found TensorFlow: (found appropriate version "1.12.0")
-- TensorFlow-ABI is 0
-- TensorFlow-INCLUDE_DIR is /usr/local/lib/python3.4/dist-packages/tensorflow/include
-- TensorFlow-LIBRARY is /usr/local/lib/python3.4/dist-packages/tensorflow/libtensorflow_framework.so
-- [WARNING] The TensorFlow version 1.12.0 has a bug (see #22766). We disable asserts using -DNDEBUG=True 
-- No TensorFlow-CC-LIBRARY detected
-- TensorFlow-SOURCE-DIRECTORY is /usr/local/include
-- Found TENSORFLOW: /usr/local/lib/python3.4/dist-packages/tensorflow/libtensorflow_framework.so (found version "1.12.0") 
CMake Error at /home/tiagoentrenamiento/Proyectos/tensorflow-cmake/cmake/modules/FindTensorFlow.cmake:222 (message):
  Project requires libtensorflow_cc.so, please specify the path in ENV-VAR
  'TENSORFLOW_BUILD_DIR'
Call Stack (most recent call first):
  CMakeLists.txt:13 (TensorFlow_REQUIRE_C_LIBRARY)

-- Configuring incomplete, errors occurred!
```

**CMakeLists:**
```
cmake_minimum_required(VERSION 2.8)
add_executable(example_1 main.cpp)
find_package(YARP REQUIRED)  # Nombre de la librer�a, buscar� un m�dulo de CMake (que define las variables utilizadas a continuaci�n) con este nombre. REQUIRED indica que lo exigimos.
find_package(OpenCV REQUIRED)  # Nombre de la librer�a, buscar� un m�dulo de CMake (que define las variables utilizadas a continuaci�n) con este nombre. REQUIRED indica que lo exigimos.
target_link_libraries(example_1 ${OpenCV_LIBS})
target_link_libraries(example_1 ${YARP_LIBRARIES})

list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/../cmake/modules)
find_package(TensorFlow 1.12.01 EXACT REQUIRED)

set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -D_GLIBCXX_USE_CXX11_ABI=${TensorFlow_ABI}")
set(CMAKE_EXE_LINKER_FLAGS  "${CMAKE_EXE_LINKER_FLAGS} -D_GLIBCXX_USE_CXX11_ABI=${TensorFlow_ABI}" )

TensorFlow_REQUIRE_C_LIBRARY()
TensorFlow_REQUIRE_SOURCE()

target_include_directories(example_1 PRIVATE TensorFlow_DEP)
target_link_libraries(example_1 PRIVATE TensorFlow_DEP)
```

davidvelascogarcia, 2019-01-25:
I`m trying to use bazel to make the TensorFlow package builder with CPU-only support. After 8553 elements get the next error:

```
ERROR: /home/tiagoentrenamiento/repos/tensorflow/tensorflow/BUILD:573:1: Executing genrule //tensorflow:tf_python_api_gen_v1 failed (Exit 1)
Traceback (most recent call last):
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/tools/api/generator/create_python_api.py", line 27, in <module>
    from tensorflow.python.tools.api.generator import doc_srcs
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/__init__.py", line 63, in <module>
    from tensorflow.python.framework.framework_lib import *  # pylint: disable=redefined-builtin
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/framework/framework_lib.py", line 52, in <module>
    from tensorflow.python.framework.importer import import_graph_def
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/framework/importer.py", line 28, in <module>
    from tensorflow.python.framework import function
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/framework/function.py", line 36, in <module>
    from tensorflow.python.ops import resource_variable_ops
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/ops/resource_variable_ops.py", line 39, in <module>
    from tensorflow.python.ops import variables
  File "/home/tiagoentrenamiento/.cache/bazel/_bazel_tiagoentrenamiento/57dc1660f9cacf8564f0a3c15ef0814f/execroot/org_tensorflow/bazel-out/host/bin/tensorflow/create_tensorflow.python_api_1_tf_python_api_gen_v1.runfiles/org_tensorflow/tensorflow/python/ops/variables.py", line 20, in <module>
    import enum  # pylint: disable=g-bad-import-order
ImportError: No module named enum
Target //tensorflow/tools/pip_package:build_pip_package failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 0.806s, Critical Path: 0.40s
INFO: 0 processes.
FAILED: Build did NOT complete successfully
```

PeterBowman, 2019-01-25:
> ImportError: No module named enum

Try https://stackoverflow.com/q/26828206.

PeterBowman, 2019-01-25:
> i have a problem looking for _libtensorflow_cc.so_

You'll probably have to build TensorFlow from source to get the C++ backend, see https://stackoverflow.com/a/33622489. The [FloopCZ/tensorflow_cc](https://github.com/FloopCZ/tensorflow_cc) project looks a bit more promising than the one I linked to in my earlier posts.

jgvictores, 2019-01-25:
1) Better to put such issues on https://github.com/roboticslab-uc3m/installation-guides so we can improve https://github.com/roboticslab-uc3m/installation-guides/blob/master/install-tensorflow.md rendered at http://robots.uc3m.es/gitbook-installation-guides/install-tensorflow.html
2) Better comments in English rather than Spanish
3) For the specific error, please use the `apt` command suggested at https://github.com/roboticslab-uc3m/installation-guides/blob/235d2a9d3498d752c7a7ee33b618c4f7a48b312f/install-tensorflow.md#dependencies-for-building-tensorflow

davidvelascogarcia, 2019-01-28:
@PeterBowman Thanks, I´m trying to use [FloopCZ/tensorflow_cc](https://github.com/FloopCZ/tensorflow_cc) project. Using `cmake`  it´s more intuitive. 
@jgvictores Thanks, it solved the problem.
>For the specific error, please use the apt command suggested at https://github.com/roboticslab-uc3m/installation-guides/blob/235d2a9d3498d752c7a7ee33b618c4f7a48b312f/install-tensorflow.md#dependencies-for-building-tensorflow

Do i open issue on [installation-guides](https://github.com/roboticslab-uc3m/installation-guides)? 
Using [FloopCZ/tensorflow_cc](https://github.com/FloopCZ/tensorflow_cc) project i can build but i get the error `line 18 not founded`

jgvictores, 2019-01-28:
1. `Thanks, it solved the problem.` > No new issue for `installation-guides`>`tensorflow`
2. `Using FloopCZ/tensorflow_cc project i can build but i get the error line 18 not founded at the end.` > new issue at https://github.com/roboticslab-uc3m/installation-guides/issues specifying tensorflow_cc in subject (guidelines [HERE](http://robots.uc3m.es/gitbook-developer-manual/best-practices-github.html#github-issues))

davidvelascogarcia, 2019-02-12:
P.R done at [tensorflowDetection2D](https://github.com/roboticslab-uc3m/vision/pull/93) #93.

jgvictores, 2019-04-10:
Some places where we've done conversions:
- https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/420d2bca11a91a34cb4e30374f6fbd730815c1ef/libraries/YarpPlugins/YarpOpenraveRGBDSensor/IRGBDSensorImpl.cpp#L59
- https://github.com/roboticslab-uc3m/openrave-yarp-plugins/blob/420d2bca11a91a34cb4e30374f6fbd730815c1ef/libraries/YarpPlugins/YarpOpenraveGrabber/IFrameGrabberImageImpl.cpp#L30-L50

davidvelascogarcia, 2019-04-10:
@jgvictores Thanks

davidvelascogarcia, 2019-04-29:
Attached: Detection example picture.

[image]

Figure 1. Detection example picture

davidvelascogarcia, 2019-09-25:
Merge done at [vision](https://github.com/roboticslab-uc3m/vision), [PR 93](https://github.com/roboticslab-uc3m/vision/pull/93).
Should we close the issue?

PeterBowman, 2019-09-25:
Sure. I've started https://github.com/roboticslab-uc3m/vision/issues/97 to cover the Travis CI stuff.

closed 2019-09-25 by PeterBowman