roboticslab-uc3m/vision #97: Continuous integration and tensorflow_cc

PeterBowman, 2019-09-25:
PR https://github.com/roboticslab-uc3m/vision/pull/93 lacked CI support on Travis due to build timeouts (https://github.com/roboticslab-uc3m/vision/pull/93#issuecomment-465135839), it takes way too long to compile this beast from sources.

Perhaps this is the perfect scenario for docker (https://github.com/roboticslab-uc3m/vision/pull/93#issuecomment-485287192 and next comments). If we could prepare a docker image in which every dependency, including tensorflow_cc, is installed and then pulled by Travis, build times should be considerably shorter. Use https://hub.docker.com/u/roboticslabuc3m to host said images (or docker run files, I'm still a docker rookie).

davidvelascogarcia, 2019-12-12:
Maybe using [tensorflow_cc](https://hub.docker.com/r/floopcz/tensorflow_cc/) `docker` image.
I´ve been doing some tests in [tensorflowDetection2D](https://github.com/davidvelascogarcia/tensorflowDetection2D) repo using `docker`image. The configuration that i used was [.travis.yml](https://github.com/davidvelascogarcia/tensorflowDetection2D/blob/develop/.travis.yml). 
Using that configuration the script build and install `yarp` from source and install `openCV`. Required time take around 15-20 minutes by commit. Attached [travis CI](https://travis-ci.org/davidvelascogarcia/tensorflowDetection2D) example using `docker`.

jgvictores, 2021-03-17:
Dropping as `wontfix` in favor of https://github.com/roboticslab-uc3m/vision/issues/103, right @PeterBowman ?

PeterBowman, 2021-03-17:
I think we should keep docker on our radar because of GPU/DNN-ready OpenCV 4.x, but yes, this issue should be dropped as we aim to supersede tensorflowDetection2D. Closing!

closed 2022-06-04 by PeterBowman