roboticslab-uc3m/vision #103: Perform object detection/classification with OpenCV/DNN

PeterBowman, 2021-01-13:
It is best to read this article found by @jgvictores: https://www.pyimagesearch.com/2020/02/03/how-to-use-opencvs-dnn-module-with-nvidia-gpus-cuda-and-cudnn/.

Also, this could complement/replace the [*tensorflowDetection2D* app](https://github.com/roboticslab-uc3m/vision/tree/01ce8ad6325198f48c3ada7de3462f0591d07872/programs/tensorflowDetection2D) (https://github.com/roboticslab-uc3m/vision/issues/92), which is somewhat lingering in an incomplete state because of https://github.com/roboticslab-uc3m/vision/issues/79. Note Tensorflow is kinda heavy-weight and does not cope well with CMake out-of-the-box.

elisabeth-ms, 2021-02-25:
I have tried dnn module from OpenCv 4.2 version with an Nvidia 940Mx (CUDA_ARCH_BIN = 5.0), although DNN from this version of OpenCv does not allow CUDA_ARCH_BIN lower than 5.3.  

The following cmake error appears “CMake Error at modules/dnn/CMakeLists.txt:99 (message):

CUDA backend for DNN module requires CC 5.3 or higher. Please remove
unsupported architectures from CUDA_ARCH_BIN option.” 

More architectures are supported in OpenCV 4.3( CUDA_ARCH_BIN>3.0). I'm going to try the dnn module with this OpenCv version.

elisabeth-ms, 2021-03-09:
Dnn module from  OpenCV 4.3 works fine with the Nvidia 940 Mx. I have tested the following example that performs object detection [https://github.com/opencv/opencv/blob/master/samples/dnn/object_detection.cpp]( https://github.com/opencv/opencv/blob/master/samples/dnn/object_detection.cpp). By default this example doesn't allow cuda backend computations but this is solved by just changing two lines of code(152-153) to:
net.setPreferableBackend(cv::dnn::DNN_BACKEND_CUDA);
net.setPreferableTarget(cv::dnn::DNN_TARGET_CUDA);

I have tested it with yolov3 weights and config from [darknet](https://pjreddie.com/darknet/yolo/) obtaining the same results when using darknet implementation in the image of the dog:

[image]

For what I have seen in the code, in addition to .cfg (Darknet) models, this example also allows to use .prototxt (Caffe), .pbtxt (TensorFlow), .xml (OpenVINO). I think that it looks like a good starting point for the yarp implementation.

elisabeth-ms, 2021-04-05:
I've developped a first version of the object detection program in [vision](https://github.com/elisabeth-ms/vision) fork (brach: object-detection-opencv-dnn). The detections (category, bounding box and confidence) are published in a yarp port. The output image with bounding boxes  annotated is also published in a yarp port.

An example of the detections image:

[image]

and the correspondant detections' data is published through the yarp port:

((brx 412) (bry 404) (category rice) (confidence 0.941261240386962890625) (tlx 317) (tly 287)) ((brx 218) (bry 431) (category milk-bottle) (confidence 0.999801039695739746094) (tlx 126) (tly 297)) ((brx 636) (bry 448) (category glass) (confidence 0.957114248275756835938) (tlx 572) (tly 344)) ((brx 496) (bry 328) (category cup) (confidence 0.999029457569122314453) (tlx 451) (tly 267))

PeterBowman, 2021-04-22:
[image]

```
((brx 442) (bry 320) (category "teddy bear") (confidence 0.241799876093864440918) (tlx 162) (tly 32))
```

Nailed it!

PS using YOLOv3-tiny model and config from the ["Performance on the COCO Dataset" section](https://pjreddie.com/darknet/yolo/) (direct links: [cfg](https://github.com/pjreddie/darknet/blob/17d47744a290a6d95f9f4d592e8955b5baa8615f/cfg/yolov3-tiny.cfg), [weights](https://pjreddie.com/media/files/yolov3-tiny.weights), [classes](https://github.com/elisabeth-ms/vision/blob/d908d5e561687d9922841fe852c2b36a7785bf0a/share/opencvDnnObjectDetection2D/conf/object_detection_classes_yolov3.txt)).

PeterBowman, 2021-04-23:
Merged at https://github.com/roboticslab-uc3m/vision/commit/4ccebfc5b42536e1d1ff63c2725c834cf88087cd, thanks! As discussed with @jgvictores, the device has been renamed to *OpencvDnnDetector*. A follow-up on the current state of *rgbDetection* will be covered in https://github.com/roboticslab-uc3m/vision/issues/104.

closed 2021-04-23 by PeterBowman