roboticslab-uc3m/vision #108: Introducing a HOG-based face detector

PeterBowman, 2022-04-21:
Our new member @cllamasmar (Carlos) is working on a face detector based on [dlib](http://dlib.net/) and the [histogram of oriented gradients](https://en.wikipedia.org/wiki/Histogram_of_oriented_gradients) algorithm. [This Python example](https://pyimagesearch.com/2021/04/19/face-detection-with-dlib-hog-and-cnn/) is probably close to the working implementation he showed me today. I have suggested that a new `IDetector` C++ component can be derived from it and reused even in Python apps. Proposed YARP device name: **HogFaceDetector**.

PS It gets funny once you know what a "hog" is (no, I didn't [check it](https://www.collinsdictionary.com/dictionary/english/hog) this morning). I hope we can stick to this name, @jgvictores.

jgvictores, 2022-04-21:
Afaik, a detector based on HOG features should/could be more generic (similar to how Haar which we use for faces, but depends on a `.xml` cascade/config file).

❤️ **HogFaceDetector**, but maybe **HogDetector** would be better?

PeterBowman, 2022-04-21:
That would be ideal. I think Carlos tried the [`get_frontal_face_detector`](http://dlib.net/imaging.html#get_frontal_face_detector) function as all he needs is detection of faces and a certain number of landmarks on them (see [example](http://dlib.net/webcam_face_pose_ex.cpp.html)). There is a more generic [object_detector](http://dlib.net/imaging.html#object_detector) class we would need to investigate further. For now, I'd go on with faces only and upgrade to generic objects when we learn more about this library. Furthermore, there are CNN utilities specifically related to face detection (check the `cnn_face_detection_model_v1` function); not sure if only available to Python code, though.

PeterBowman, 2022-04-21:
ASWJ, we would rather explore a generic HOG implementation with OpenCV. It is a better option than dlib from our point of view since there is no need to configure another dependency. Also, we have no experience with dlib.

tl;dr: looking forward to detecting hogs in OpenCV, but no plans to do so in the short-mid-long term, therefore closing as invalid/a-bit-too-rushed-my-bad. (@cllamasmar please re-open if you would like to work on this!)

closed 2022-04-21 by PeterBowman

PeterBowman, 2022-05-12:
> looking forward to detecting hogs in OpenCV, but no plans to do so in the short-mid-long term

We are currently revisiting this.

PeterBowman, 2022-05-24:
Important to note: dlib must be compiled and installed from sources (https://github.com/davisking/dlib) instead of via apt (`libdlib-dev` package) due to a bug related to `dlib::cv_image` which prevents from compiling dependent projects.

cllamasmar, 2022-06-02:
https://medium.com/codex/facial-landmark-detection-algorithms-5b2d2a12adaf
https://github.com/opencv/opencv_contrib/tree/b2904b94414e885a595ec0c56fa70da5a3fe8472/modules/face/src

cllamasmar, 2022-06-02:
[image]