roboticslab-uc3m/vision #111: Add ArUco detector

PeterBowman, 2024-06-06:
Available since OpenCV 4.7.0:
- module: objdetect
- https://docs.opencv.org/4.10.0/d5/dae/tutorial_aruco_detection.html
- [`cv::aruco::ArucoDetector`](https://docs.opencv.org/4.10.0/d2/d1a/classcv_1_1aruco_1_1ArucoDetector.html)

cc @anacg1620 @jgvictores

PeterBowman, 2024-06-07:
@anacg1620 I have somewhat circumvented the mismatching hash issue, you might want to merge the master branch into yours or rebase on it.

anacg1620, 2024-06-17:
Done as of https://github.com/roboticslab-uc3m/vision/commit/a19c99ff3478797b61d4a7662a98ed9d20f06d3a

closed 2024-06-17 by anacg1620

PeterBowman, 2024-06-18:
@jgvictores since we have `QrDetector` instead of `QrCodeDetector`, perhaps this component should be renamed to `ArucoDetector`?

PeterBowman, 2024-06-18:
> @jgvictores since we have `QrDetector` instead of `QrCodeDetector`, perhaps this component should be renamed to `ArucoDetector`?

Done at https://github.com/roboticslab-uc3m/vision/commit/1cb8da25971efa227220ff50d09d24912bcf9c1a.

PeterBowman, 2024-07-23:
Tests and C++/Python examples added by @anacg1620 at https://github.com/roboticslab-uc3m/vision/commit/8fb126f9c1fbff12995c1c215f02754755e7af1c. Thanks!